Ultimaker/Cura #1574: fatal exception Cura 2.5 beta

mrveedee, 2017-03-24:
Hello,

I cannot start up the newest Cura (2.5 beta). I am working with Windows 10 64 bit.
When I launch the program, u will see him loading machines but after this he will give me this error:

"Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.5/build/inst/bin/cura_app.py", line 71, in <module>
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 586, in run
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 101, in __init__
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 630, in getMachineManager
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 1116, in createMachineManager
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 75, in __init__
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 342, in setActiveMachine
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Application.py", line 145, in setGlobalContainerStack
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 201, in emit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 435, in _onStackChanged
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 549, in _updateDisallowedAreas
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 583, in _computeDisallowedAreasPrinted
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 519, in getResolveOrValue
AttributeError: 'NoneType' object has no attribute 'hasProperty'
"

After a few seconds another fatal exception:
"Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 652, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 216, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Event.py", line 187, in call
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 435, in _onStackChanged
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 549, in _updateDisallowedAreas
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 583, in _computeDisallowedAreasPrinted
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 519, in getResolveOrValue
AttributeError: 'NoneType' object has no attribute 'hasProperty'
"

What could be the problem here? Thanks for the help.

Ghostkeeper, 2017-03-27:
Hmm, it's not finding any "user changes" or "quality changes" element on the extruder stack... In less technical words, it's not finding either the current changes or a custom profile that was supposed to be attached to the printer you added.

You could try removing/renaming your settings folder. It's in `C:\Users\<your Windows username>\AppData\Local\cura`. If you rename that folder to `cura-corrupt` you'll have reset Cura to factory settings and it should no longer crash on start-up. However, you'd lose your settings.

In fact, could you perhaps put that folder in a .zip and attach that .zip here or make it available for us to download? That could help in debugging this.

mrveedee, 2017-03-29:
Hello Ghostkeeper,

Thank you for your reply. I do not find a settings folder inside appdata\local\cura.
Of course, below you will find the ZIP of the entire map.

[cura.zip](https://github.com/Ultimaker/Cura/files/878463/cura.zip)

Ghostkeeper, 2017-03-29:
That zip file is giving me errors when I extract it :(

closed 2017-07-27 by nallath