Ultimaker/Cura #1575: Fatal Exception 2.5.0 BETA

SwiftyJ, 2017-03-24:
This occurred when I was changing material diameter and it was set to 0.

Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "C:\Program Files\Cura 2.5\plugins\CuraEngineBackend\CuraEngineBackend.py", line 190, in slice
    self.printDurationMessage.emit(0, [0])
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 201, in emit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\PrintInformation.py", line 122, in _onPrintDurationMessage
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\PrintInformation.py", line 162, in _calculateInformation
ZeroDivisionError: float division by zero

nallath, 2017-03-27:
For our reference; CURA-3594

jackha, 2017-03-29:
I couldn't reproduce, but the cause has been taken away (https://github.com/Ultimaker/Cura/pull/1586). Thanks for reporting.

fieldOfView, 2017-03-29:
Can confirm this is a duplicate of #1582, which was fixed. This issue can be closed.

closed 2017-03-29 by awhiemstra