Ultimaker/Cura #1577: Fatal Exception 2.5.0 BETA

ChrisRiddell, 2017-03-25:
I seen there is already to reports of this but with different messages so might be unrelated?

This came up when I tried to drag and drop a STL into cura.

> Version: 2.5.0-BETA
Platform: Darwin-16.4.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1
Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 652, in event
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 216, in event
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Event.py", line 187, in call
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 305, in __performEmit
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 124, in calculateBoundingBoxMesh
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 581, in getBoundingBox
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 648, in _calculateAABB
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 176, in getExtents
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 251, in getConvexHullVertices
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 243, in getConvexHull
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 235, in _computeConvexHull
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 365, in approximateConvexHull
  File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
  File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
scipy.spatial.qhull.QhullError: Qhull error

ChrisRiddell, 2017-03-25:
Want to note so far it's only happening on this STL

http://www.thingiverse.com/thing:1828238

Appesteijn, 2017-03-25:
Can confirm, also crashes on my Mac

Vandrasc, 2017-03-27:
Same behaviour on Win10 64bit

nallath, 2017-03-27:
I think we already have an issue for this, but it was moved to a lower priority as we had difficulty reproducing it.

ChrisRiddell, 2017-03-27:
I have tested on a couple computers both Mac and Windows and that linked file from my first comment crashes cura every time.

This is the full error that is produced to the cura.log file

> 2017-03-27 18:15:01,832 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [408]: Calculating normals took 0.00015997886657714844 seconds
2017-03-27 18:15:01,834 - DEBUG - STLReader.STLReader.read [63]: Loaded a mesh with 321 vertices
2017-03-27 18:15:01,837 - ERROR - UM.Logger.logException [74]: Exception: Exception occurred while loading file /Users/chrisriddell/Downloads/Ultimaker_2_Print_Head_Axis_DisAssemble_Tool/files/UM_Axes_Tool.stl
2017-03-27 18:15:01,840 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-03-27 18:15:01,842 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/FileHandler/ReadFileJob.py", line 61, in run
2017-03-27 18:15:01,844 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshFileHandler.py", line 37, in readerRead
2017-03-27 18:15:01,846 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 176, in getExtents
2017-03-27 18:15:01,847 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 251, in getConvexHullVertices
2017-03-27 18:15:01,850 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 243, in getConvexHull
2017-03-27 18:15:01,852 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 235, in _computeConvexHull
2017-03-27 18:15:01,854 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 365, in approximateConvexHull
2017-03-27 18:15:01,856 - ERROR - UM.Logger.logException [78]:   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
2017-03-27 18:15:01,859 - ERROR - UM.Logger.logException [78]:   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
2017-03-27 18:15:01,861 - ERROR - UM.Logger.logException [78]: scipy.spatial.qhull.QhullError: Qhull error
2017-03-27 18:15:01,901 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-03-27 18:15:01,905 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-03-27 18:15:01,906 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 652, in event
2017-03-27 18:15:01,908 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 216, in event
2017-03-27 18:15:01,910 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Event.py", line 187, in call
2017-03-27 18:15:01,912 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 305, in __performEmit
2017-03-27 18:15:01,914 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 124, in calculateBoundingBoxMesh
2017-03-27 18:15:01,916 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 581, in getBoundingBox
2017-03-27 18:15:01,917 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 648, in _calculateAABB
2017-03-27 18:15:01,918 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 176, in getExtents
2017-03-27 18:15:01,920 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 251, in getConvexHullVertices
2017-03-27 18:15:01,921 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 243, in getConvexHull
2017-03-27 18:15:01,923 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 235, in _computeConvexHull
2017-03-27 18:15:01,925 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 365, in approximateConvexHull
2017-03-27 18:15:01,926 - CRITICAL - cura.CrashHandler.show [39]:   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
2017-03-27 18:15:01,928 - CRITICAL - cura.CrashHandler.show [39]:   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
2017-03-27 18:15:01,929 - CRITICAL - cura.CrashHandler.show [39]: scipy.spatial.qhull.QhullError: Qhull error

Ghostkeeper, 2017-03-27:
Doesn't seem to do anything wrong on Ubuntu. At least not for me. Tried both 2.5 and master. We might have a platform-specific issue here then.

ChrisRiddell, 2017-07-06:
This issue seems to be fixed in 2.6.1 the file linked above no loner crashes Cura.

closed 2017-07-06 by nallath