Ultimaker/Cura #1579: Problems with the start gcode.

Toriniasty, 2017-03-25:
Hi,
So I'm having trouble with setting up following start gcode.
```
;PROFILE SETTINGS
;NOZZLE DIAMETER: {machine_nozzle_size}
;FILAMENT DIAMETER: {material_diameter}
;PRINT SPEED: {speed_print}
;PRINT TRAVEL: {speed_travel}
;MATERIAL FLOW: {material_flow}
;LAYER HEIGHT: {layer_height}
;1ST LAYER HEIGHT: {layer_height_0}
;TOP BOTTOM THICKNESS: {top_bottom_thickness}
;BOTTOM LAYERS: {bottom_layers}
;LINE WIDTH: {line_width}
;WALL LINE COUNT: {wall_line_count}
;WALL THICKNESS: {wall_thickness}
;INFILL DENSITY: {infill_sparse_density}
;INFILL PATTERN: {infill_pattern}
;PRINT TEMP: {material_print_temperature}
;BED TEMP: {material_bed_temperature}
;ADHESION TYPE: {adhesion_type}
;SKIRT LINES: {skirt_line_count}
;BRIM WIDTH: {brim_width}
;SUPPORT?: {support_enable}
;SUPPORT TYPE: {support_type}
;SUPPORT XY: {support_xy_distance}
;SUPPORT Z: {support_z_distance}
;SUPPORT DENSITY: {support_infill_rate}
;SUPPORT PATTERN: {support_pattern}
;RETRACTION DISTANCE: {retraction_amount}
;RETRACTION SPEED: {retraction_speed}
;COOLING?: {cool_fan_enabled}

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28
G29
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F9000
```

After putting it in the menu, it doesn't take it in consideration and after cura restart drops it to:
```;PROFILE SETTINGS

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28
G29
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F9000
```

However in the: Anet+A8_settings.inst.cfg I can see:
```
[general]
version = 2
name = Anet A8_settings
definition = custom

[metadata]
type = definition_changes

[values]
gantry_height = 0
machine_depth = 220
machine_end_gcode = ;End GCode
	M104 S0                     ;extruder heater off
	M140 S0                     ;heated bed heater off (if you have it)
	G91                                    ;relative positioning
	G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
	G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
	G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
	M84                         ;steppers off
	G90                         ;absolute positioning
	G1 X0 Y180 F9000            ;move forward
machine_head_with_fans_polygon = [[0, 0], [0, 0], [0, 0], [0, 0]]
machine_heated_bed = True
machine_height = 220
machine_start_gcode = ;PROFILE SETTINGS
	;NOZZLE DIAMETER: {machine_nozzle_size}
	;FILAMENT DIAMETER: {material_diameter}
	;PRINT SPEED: {speed_print}
	;PRINT TRAVEL: {speed_travel}
	;MATERIAL FLOW: {material_flow}
	;LAYER HEIGHT: {layer_height}
	;1ST LAYER HEIGHT: {layer_height_0}
	;TOP BOTTOM THICKNESS: {top_bottom_thickness}
	;BOTTOM LAYERS: {bottom_layers}
	;LINE WIDTH: {line_width}
	;WALL LINE COUNT: {wall_line_count}
	;WALL THICKNESS: {wall_thickness}
	;INFILL DENSITY: {infill_sparse_density}
	;INFILL PATTERN: {infill_pattern}
	;PRINT TEMP: {material_print_temperature}
	;BED TEMP: {material_bed_temperature}
	;ADHESION TYPE: {adhesion_type}
	;SKIRT LINES: {skirt_line_count}
	;BRIM WIDTH: {brim_width}
	;SUPPORT?: {support_enable}
	;SUPPORT TYPE: {support_type}
	;SUPPORT XY: {support_xy_distance}
	;SUPPORT Z: {support_z_distance}
	;SUPPORT DENSITY: {support_infill_rate}
	;SUPPORT PATTERN: {support_pattern}
	;RETRACTION DISTANCE: {retraction_amount}
	;RETRACTION SPEED: {retraction_speed}
	;COOLING?: {cool_fan_enabled}
	
	;GCODE
	G21 ;metric values
	G90 ;absolute positioning
	M82 ;set extruder to absolute mode
	M107 ;start with the fan off
	G28 X0 Y0 ;move X/Y to min endstops
	G28
	G29
	G1 Z15.0 F9000 ;move the platform down 15mm
	G92 E0 ;zero the extruded length
	G1 F9000
machine_width = 220
```

So I've tried to clean start cura, and moved old cura directory in appdata/local but it behaves the same way.

Resulting gcode doesn't contain profile settings.

Can you help me with that? Did I do something that I shouldn't?

Cheers

fieldOfView, 2017-03-25:
There is a known issue that lines that start with a ; are not stored.
Duplicate of https://github.com/Ultimaker/Cura/issues/1237

fieldOfView, 2017-03-25:
Note that profiles are already stored at the bottom of the gcode file.

Toriniasty, 2017-03-25:
Awesome. Thanks for quick reply! I'll close the issue as it's well known, no point of having it opened.

closed 2017-03-25 by Toriniasty