Ultimaker/Cura #1580: 2.5.0 BETA does not load my plugin anymore

probonopd, 2017-03-25:
It is working in 2.4 but not in 2.5.0 BETA:

```
me@host:~$ ./Downloads/Cura-2.5.0-BETA.AppImage 2>&1 | grep ERROR
2017-03-25 16:51:51,723 - ERROR - UM.PluginRegistry.loadPlugin [91]: Plugin VersionUpgrade24to25 did not return any objects to register
2017-03-25 16:51:51,785 - ERROR - UM.Logger.logException [74]: Exception: Error loading plugin WirelessPrinting:
2017-03-25 16:51:51,787 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-03-25 16:51:51,788 - ERROR - UM.Logger.logException [78]:   File "/build/2.5/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py", line 89, in loadPlugin
2017-03-25 16:51:51,790 - ERROR - UM.Logger.logException [78]:   File "/home/me/.local/share/cura/plugins/WirelessPrinting/__init__.py", line 21, in register
2017-03-25 16:51:51,791 - ERROR - UM.Logger.logException [78]:     "machine_action": DiscoverWirelessPrintAction.DiscoverWirelessPrintAction()
2017-03-25 16:51:51,793 - ERROR - UM.Logger.logException [78]:   File "/home/me/.local/share/cura/plugins/WirelessPrinting/DiscoverWirelessPrintAction.py", line 29, in __init__
2017-03-25 16:51:51,794 - ERROR - UM.Logger.logException [78]:     cura.Settings.CuraContainerRegistry.getInstance().containerAdded.connect(self._onContainerAdded)
2017-03-25 16:51:51,796 - ERROR - UM.Logger.logException [78]: AttributeError: module 'cura.Settings.CuraContainerRegistry' has no attribute 'getInstance'
```

Was something changed that I need to take care of in the plugin?
(Cross-reference: https://github.com/probonopd/WirelessPrinting/issues/13)

fieldOfView, 2017-03-25:
Yes, you need to make a minor change. See eg here:
https://github.com/fieldOfView/OctoPrintPlugin/commit/fd8bb1e6f7dd46f01d2fa028d5dae9031c99052c

fieldOfView, 2017-03-25:
It is good practice to make branches for your plugin to match the branch of Cura. APIs may/will change between versions of Cura.

probonopd, 2017-03-25:
Thank you @fieldOfView

closed 2017-03-25 by probonopd