Ultimaker/Cura #1581: Errant M104 S0 near the end of layer 0 when slicing temp set to 0 (for sending g-code later)

SirGeekALot, 2017-03-25:
OS: Windows 10 Pro 64-bit.
Cura version: Cura-2.4.0-win64.exe
3D printer: reprap running Marlin on a RAMPs 1.4

Settings:
Heated bed: yes
Extruder temp: 0
Heated bed temp: 0

I usually slice my models in Cura, then save the g-code.  Sometimes I do it all in Cura (if I'm on the laptop that host the 3D printer), but usually I slice on my fast machine, then save the g-code file and send it to my laptop 3D printer host, where I then use Pronterface to send it.

Repro Steps:
I opened the file in Pronterface, connected to the printer, set its extruder temp to 210 (for PLA) and waited.
I manually primed the extruder a few times, then I started printing.

Result:
The first layer printed fine, except at what appeared to be the end of the first layer (it was), I started seeing "cold extrusion prevented" errors.  Since this was my first attempt at using Cura 2 (instead of Cura 15.04.6) and nothing else had changed, I suspected the new Cura.  I sliced the same STL file with the older Cura, using nearly identical settings, and it worked flawlessly.

Cause:
Looking at the gcode file, I found an "M104 S0" command on line 9983 (about 37 lines before the end of Layer 0).

Supporting files:
Attached STL file, Cura 2 generated gcode file, and two screenshots showing my Cura 2 slicing and machine settings, plus the line of code where

[C_iPhone6SPlusCase_v4_reproTest1.zip](https://github.com/Ultimaker/Cura/files/870264/C_iPhone6SPlusCase_v4_reproTest1.zip)

Ghostkeeper, 2017-03-27:
You've set the extruder temperature to 0. Since 2.4 we've had a separate temperature for the initial layer, so before the start of the second layer it'll set it to the normal printing temperature, which is 0 in your case.

In 2.5 we fixed a related thing where the description of the temperature said that setting a temperature to 0 won't add any temperature commands but commands were added.

SirGeekALot, 2017-03-27:
Ok, so the fix is for me to wait for v2.5 before attemptng to use Cura 2.x.  Got it.

closed 2017-03-27 by SirGeekALot

BagelOrb, 2017-03-28:
You can set the initial layer temperature at zero as well. I think that prevents the M104 S0 from appearing.