Ultimaker/Cura #1582: can't change filiment dia

thomllama, 2017-03-25:
got fatal error as soon as I tried to change filament dia to 1.75

Version: 2.5.0-BETA
Platform: Darwin-16.4.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/Applications/Cura 2.5.app/Contents/MacOS/plugins/plugins/CuraEngineBackend/CuraEngineBackend.py", line 190, in slice
    self.printDurationMessage.emit(0, [0])
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 201, in emit
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 305, in __performEmit
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/cura/PrintInformation.py", line 122, in _onPrintDurationMessage
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/cura/PrintInformation.py", line 162, in _calculateInformation
ZeroDivisionError: float division by zero

smartavionics, 2017-03-26:
Deleting all the digits probably sets the value to 0 so as a workaround, pause slicing before changing the dia and un-pause after you have set 1.75.

thomllama, 2017-03-26:
wouldn't let me even start it up.  crashed 2.4 also. had to delete and use a cleaner to remove all files then reinstall. Installed 2.5 all alone and it seems to be working fine now. 

Do wish they would start having seperate prefs/settings folders especially for beta versions so issues wouldn't cross bleed to the stable versions.

fieldOfView, 2017-03-26:
> Do wish they would start having seperate prefs/settings folders especially for beta versions so issues wouldn't cross bleed to the stable versions.

That is in the works for Cura 2.6

thomllama, 2017-03-26:
SWEEETTTT!!!  :P

thomllama, 2017-03-26:
as a note, since deleting everything and installing 2.5 alone, seems to be working very well. 
Thanks for all the hard work folks!!

fieldOfView, 2017-03-26:
I have also made a fix for your issue, so it should be fixed before 2.5 goes out of beta. Thanks for the report.
https://github.com/Ultimaker/Cura/issues/1586

thomllama, 2017-03-26:
Cool. :)

closed 2017-03-29 by jackha