Ultimaker/Cura #1583: Slicing freezes Cura 2.5 BETA

ChrisRiddell, 2017-03-26:
When slicing this 

On OSX 10.12.3

the whole of Cura 2.5 Beta freezes becomes unresponsive for about 5 seconds before the slice bar starts moving, when you look at the log there is over 7000 lines of this message 

> 2017-03-26 10:27:14,668 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186809.
2017-03-26 10:27:14,670 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186813.
2017-03-26 10:27:14,672 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186819.
2017-03-26 10:27:14,673 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186830.
2017-03-26 10:27:14,674 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186833.
2017-03-26 10:27:14,676 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186835.
2017-03-26 10:27:14,677 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186838.
2017-03-26 10:27:14,678 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186850.
2017-03-26 10:27:14,679 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] Couldn't find face connected to face 186862.

nallath, 2017-03-27:
@BagelOrb; I think we should remove this logging, as it seems to be slowing down the GUI. What are your thoughts?

fieldOfView, 2017-03-27:
Perhaps keep a count and print the number of occurences?

BagelOrb, 2017-03-27:
(Internal issue: CURA-3601)

BagelOrb, 2017-03-27:
fixed

BagelOrb, 2017-03-27:
Didn't keep a count, though....

nallath, 2017-03-30:
This should be fixed for the 2.5 release.

closed 2017-03-30 by nallath