Ultimaker/Cura #1584: Cura updater

Austinthemighty, 2017-03-26:
We need a way of checking and updating cura from inside cura itself.  It is a pain in the ass to go to ultimakers website and download, uninstall and reinstall cura ever time there is an update.

fieldOfView, 2017-03-26:
That would be nice, but also non-trivial. Updating executables while they are running can be tricky, depending on the platform.

Ghostkeeper, 2017-03-27:
On Windows it could download the installer and run it. It'll ask the user to close Cura in the initial pages of the installer's wizard.

On Linux, there's the matter of where the AppImage file should be placed. If the file is placed it could theoretically close itself and start the new application.

On MacOS there's a similar problem of Linux but the location is clear (Applications). However the name of the file isn't. There's also the matter of file rights (also on Linux).

probonopd, 2017-03-27:
Check [AppImageUpdate](https://github.com/probonopd/AppImageUpdate) for delta binary updates (i.e., only the bytes that have actually changed need to be downloaded). `linuxdeployqt` has a command line option to embed the necessary update information into the AppImage file. Let me know if you are interested and need more information.

nallath, 2017-03-28:
I think it would be best to have an uniform solution for all operating systems. Otherwise we will have to design multiple systems, which will lead to a lot of overhead.

fieldOfView, 2017-03-28:
If that means we can skip the survey, I am all for it ;)

UltimateSports, 2017-04-05:
Even if it were a checker system that would inform you when there was an updated version on the website to download! too often i will continue using the one I have now for weeks after there is something new on the website.

nallath, 2017-04-05:
That's because we tend to forget to update the number :$

Ghostkeeper, 2017-04-06:
There is a system that checks for updates on the sites and notifies you in Cura. I thought that works, right?

UltimateSports, 2017-04-06:
possibly, how frequently are there updates released for the beta version? as cura still says its 2.5.0, and i didn't get a message about the 2.5.1 update which i noticed on the website when i checked out of curiosity after your comment.

fieldOfView, 2017-04-06:
There is no 2.5.0 yet. There's a 2.5.0 beta and a 2.5.0 beta 2.

UltimateSports, 2017-04-06:
Sorry, I missed the bit about the beta as i assumed we were referring to the beta versions. surely numbering would make more sense if instead of using 2.5.0 _BETA 2_ using _"2.5.1 Beta" or "Beata 2.5.1"_ ?

Ghostkeeper, 2017-04-06:
We number our releases as `2.3.0`, `2.3.1`, `2.3.2`, `2.4.0`, `2.5.0`, etc...

For each of the major releases (ending in `.0`) we release a number of beta versions so that people can preview the changes and help us testing them. We just released the second beta for the `2.5.0` version, so we call it `2.5 beta 2`.

The automated update checker doesn't check for beta versions. They are not meant to be entirely stable.

awhiemstra, 2017-04-06:
I'd like to include support for "update channels" at some point in the future where you can indicate that you'd like to run nightly/beta/stable versions and get update notifications for that. But it will be a while before that shows up.

> Sorry, I missed the bit about the beta as i assumed we were referring to the beta versions. surely numbering would make more sense if instead of using 2.5.0 BETA 2 using "2.5.1 Beta" or "Beata 2.5.1" ?

The current beta is the second beta release for what eventually will become the 2.5.0 release. Calling it 2.5.1 beta would mean it is the beta release for what will become the 2.5.1 version.

UltimateSports, 2017-04-06:
ok, that makes more sense, thanks for clarification. but comment about update checker would still be appreciated for beta versions if this would be a possibility in the future.

nallath, 2017-04-06:
I also forgot to update the number to 2.4. I did that yesterday, so everyone with the 2.3.1 release should get a notification.

catch22eu, 2017-04-16:
Usually not the easy way to implement from development perspective, but probably appreciated from user point of view: have applications update natively, ex.: 
OSX: App store
Debian (linux): .deb via official repository
etc.

probonopd, 2017-04-16:
> Debian (linux): .deb via official repository

This means that e.g., a Ubuntu 16.04 user would be stuck with an early 2016 version and never get a recent Cura...

catch22eu, 2017-04-16:
> This would mean that e.g., a Ubuntu 16.04 user would be stuck with an early 2016 version and never get a recent Cura... and the applications in debian stable are practically outdated the minute the distribution is released.

Package maintainers can update a package with new versions as they please, for whichever release they please. This requires the introduction of maintainers of these packages. Updating these packages can be part of the development cycle of the software. It requires some more effort when releasing new software though, and selection of which packages and distributions to support, but relieves these users from updating hassle, as the OP points out. Just a thought...

probonopd, 2017-04-16:
> Package maintainers can update a package with new versions as they please, for whichever release they please.

As far as I know, distributions such as Ubuntu and Debian don't update the versions anymore after the release, especially of LTS versions. Which means that users would have to introduce third-party backports or ppa repositories into their systems. Compared to that, I prefer the current solution much more, that is, download the latest AppImage and be done with it.

awhiemstra, 2017-04-18:
> Compared to that, I prefer the current solution much more, that is, download the latest AppImage and be done with it.

In fact, it is one of the main reasons things like AppImage and Flatpak are gaining popularity: It makes it easy to have a recent version of the software without needing to make changes to the base platform.

thopiekar, 2017-07-05:
@awhiemstra : A more recent version doesn't mean that you have always less bugs :wink: