Ultimaker/Cura #1585: Print Speed and Infill Speed set to same value causes infill speed to use outer wall speed instead

Pheneeny, 2017-03-26:
Application Version: 2.4.0
Platform: Linux Mint 18.1

Steps to Reproduce:
Slice a model with the following speed parameters:
Print Speed: 100 mm/s
Infill Speed: 100 mm/s
Outer Wall Speed: 30 mm/s
Inner Wall Speed: 60 mm/s

Look at the infill sections of the gcode produced. The feed rate is never set during infill moves, causing the infill to use that last feed rate set, which was the outer wall speed. So in this case, the infill will print at 30.

Now slice the same model with all of the same speed settings, but change:
Infill Speed: 101 mm/s

Look at the new gcode and notice the Infill has proper feed rates set, and now prints at 101 mm/s

Actual Results:
When print speed and infill speed are set to the same value, the feed rate is never specified for infill in the gcode, so it uses the last feed rate that was set, which is the outer wall speed.

Expected results:
Infill should print using the infill speed.

Ghostkeeper, 2017-03-27:
I'm thinking this might depend on which other settings you're using. What printer/profile are you using?

This is a snippet of the g-code I'm seeing:

    G1 X96.675 Y127.325 E355.19942
    M204 S5000
    M205 X30
    G0 F15000 X96.675 Y127.125
    G0 X97.212 Y126.7
    G0 X135.548 Y124.942
    ;TIME_ELAPSED:1547.912028
    ;LAYER:11
    G0 X135.548 Y124.942 Z1.37
    M204 S4000
    M205 X25
    ;TYPE:FILL
    G1 F6000 X99.057 Y88.45 E355.6039
    M204 S5000
    M205 X30
    G0 F15000 X100.533 Y88.449
    M204 S4000
    M205 X25
    G1 F6000 X97.45 Y99.954 E355.69726

As you can see, the two infill lines towards the end have `F6000` in them, because the previous move was a travel move at `F15000`. I used an Ultimaker 3 with default settings (on master).

Pheneeny, 2017-03-28:
I am attaching my current profile and two gcode outputs. same_speed_and_infill.gcode is the bugged output when infill and print speed are set the same, different_speed_and_infill.gcode is the correct output when the infill speed is increased by 1. 
[custom_bugged profile.txt](https://github.com/Ultimaker/Cura/files/874196/custom_bugged.profile.txt)
[different_speed_and_infill.gcode.txt](https://github.com/Ultimaker/Cura/files/874197/different_speed_and_infill.gcode.txt)
[same_speed_and_infill.gcode.txt](https://github.com/Ultimaker/Cura/files/874198/same_speed_and_infill.gcode.txt)

BagelOrb, 2017-03-28:
Could you attach the project perhaps?
The project contains all information.

I'd like to know what type of machine you have and whether it used custom extruder start gcode commands.

Pheneeny, 2017-03-28:
I have attached the project file. The machine is a D-Bot, the only custom gcode commands I added were to hold the temperature at 40C after printing was complete.

[bugged_project.zip](https://github.com/Ultimaker/Cura/files/875857/bugged_project.zip)

Ghostkeeper, 2017-03-29:
I'm not able to load that project due to a missing definition file (my Cura doesn't know about any SturD printer). But I am getting a general gist of your settings from that project. The model, by the way, barely has any infill. But there is some. Here is a snippet of layer 33:

    ;LAYER:33
    G0 X158.048 Y159.529 Z6.87
    M204 S4000
    M205 X25
    ;TYPE:FILL
    G1 F1500 E125.71072
    G1 F6000 X156.95 Y159.824 E125.72854
    M204 S5000
    M205 X30
    G0 X158.049 Y158.049
    M204 S4000
    M205 X25
    G1 X157.42 Y157.42 E125.74249
    M204 S5000
    M205 X30
    G0 X157.8 Y156.8
    M204 S1000
    M205 X10
    ;TYPE:WALL-INNER
    G1 F3000 X157.8 Y151.2 E125.79516

So the first infill line starts with `G1 F6000` (100mm/s). It then does a travel move at the same speed, and then another infill line at the same speed. Then another travel move towards the inner wall and the infill is done.

This is correct however because both travel speed and infill speed are set to 100mm/s in your profile. So both should be printing at F6000.

nallath, 2017-03-29:
The project file should load the definition from the project. Seems like we found a bug there.

Pheneeny, 2017-03-29:
I've uploaded a new project file with a model with some more infill, along with the two gcode outputs for when infill speed is the same as print speed and when it is different. 

SturD is just the name I gave a custom FDM printer.

[bugged_project2.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/879247/bugged_project2.curaproject.3mf.zip)
[same.gcode.txt](https://github.com/Ultimaker/Cura/files/879248/same.gcode.txt)
[different.gcode.txt](https://github.com/Ultimaker/Cura/files/879249/different.gcode.txt)

Ghostkeeper, 2017-03-30:
I'm still not seeing the problem :(

Even in your g-code files. Here's a snippet from same.gcode.txt:

    <printing skin here>
    G1 X157.191 Y154.363 E1032.48717
    G0 F6000 X157.047 Y154.219 ;<-- This one sets the speed correctly.
    ;TIME_ELAPSED:793.982264
    ;LAYER:69
    G0 X157.412 Y154.36 Z14.1
    ;TYPE:FILL
    G1 X156.211 Y158.839 E1032.64141 ;<-- Still on F6000!
    G0 X156.186 Y158.935
    G0 X158.839 Y158.839
    G1 X141.159 Y141.159 E1033.47302
    G0 X141.089 Y141.089
    G0 X142.314 Y141.159
    G1 X141.159 Y145.47 E1033.62146

Right before it starts with infill there's always a `G0 F6000` that should cause the infill to be printed at `F6000` or 100mm/s.

Pheneeny, 2017-03-31:
I've done some investigating and I believe I've found the source of the problem / confusion. Glancing at some firmware, it looks like Marlin has a single feed rate, and whether it is set in a G0 or a G1 command doesn't matter. I am using a board running Smoothieware, and Smoothieware tracks and applies feed rates from G0 and G1 separately.

So when the gcode specifies a feedrate in a G0 command, that is only applied to other G0 commands. Since my settings have the travel speed and the infill speed both set to 100mm/s, the gcode sets the feed rate to 100mm/s in a G0 command, but never sets it again for a G1 command

From the same.gcode:

```
G0 X158.781 Y154.256
G1 F1800 X158.324 Y154.364 E1032.45976 ;<-- Sets G1 speed to 1800 or 60mm/s
G1 X157.757 Y154.362 E1032.47348
G1 X157.191 Y154.363 E1032.48717
G0 F6000 X157.047 Y154.219 ; <-- Sets G0 speed to 6000 or 100mm/s
;TIME_ELAPSED:793.982264
;LAYER:69
G0 X157.412 Y154.36 Z14.1
;TYPE:FILL
G1 X156.211 Y158.839 E1032.64141 <-- G1 Speed is not set, so it is still 1800 instead of 6000
G0 X156.186 Y158.935 ; <-- G0 Speed is correctly set to 6000
G0 X158.839 Y158.839
G1 X141.159 Y141.159 E1033.47302
G0 X141.089 Y141.089
```

Compare that to the gcode from when infill speed was set to 101 instead of 100:
```
G0 X158.781 Y154.256
G1 F1800 X158.324 Y154.364 E1032.45976 ; <-- G1 speed set to 1800 for skin
G1 X157.757 Y154.362 E1032.47348
G1 X157.191 Y154.363 E1032.48717
G0 F6000 X157.047 Y154.219 ; <-- G0 speed set to 6000 for travel
;TIME_ELAPSED:793.222705
;LAYER:69
G0 X157.412 Y154.36 Z14.1
;TYPE:FILL
G1 F6060 X156.211 Y158.839 E1032.64141 ; <-- G1 speed set to 6060 for infill print
G0 X156.186 Y158.935
G0 F6000 X158.839 Y158.839 ; <-- G0 speed set back down to 6000 for travel
```
So it looks like the root cause is that Cura assumes that a feed rate being set in either G0 or G1 applies to both G0 or G1 moves, while Smoothieware tracks those separately, so a G0 feed rate only applies to  G0 commands, and a G1 feed rate only applies to G1 commands, and it is required that BOTH G0 and G1 to specify their own feed rates. 

Since Cura does not have a GCode Flavor for Smoothie, I've been using RepRap(Marlin/Sprinter). So this issue may be turning into either a request for a Smoothie GCode flavor, or a request to have Cura set both G0 and G1 speeds to their appropriate values separately.

nallath, 2017-03-31:
Both of those requests are not likely to be picked up by Ultimaker, as we only work with Marlin based firmware. But feel free to make a pull request.

fieldOfView, 2017-03-31:
It could also be solved with a postprocessing script.

nallath, 2017-03-31:
True, but that's not a very nice way of doing it. Especially because you would need to add the script every time you print something, which is easily forgotten.

Ghostkeeper, 2017-04-03:
Indeed, setting the speed to 101 instead of 100 is an easier workaround and doesn't do much harm.

To fix this properly, you'd need to add a g-code flavour to the engine and make sure that the engine tracks two flow rates in the gcodeExport. Upon writing a move, wherever the engine checks the last-written flow rate to the new flow rate to determine whether to add an F or not, it needs to check against the proper flow rate for G0 or G1 (but only do this for the new Smoothie flavour).

BagelOrb, 2017-07-21:
Please use the workaround

closed 2017-07-21 by BagelOrb