Ultimaker/Cura #1588: JSONDecodeError at the end of print

TinkerGnome, 2017-03-26:
Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1
Printer: UM3

Exception:
Traceback (most recent call last):
  File "D:\Programs\Cura 2.5\plugins\UM3NetworkPrinting\NetworkPrinterOutputDevice.py", line 923, in _onFinished
    self._json_printer_state = json.loads(bytes(reply.readAll()).decode("utf-8"))
  File "D:\env\inst\Lib\json\__init__.py", line 319, in loads
  File "D:\env\inst\Lib\json\decoder.py", line 339, in decode
  File "D:\env\inst\Lib\json\decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 1 column 1249 (char 1248)

nallath, 2017-03-27:
Uuh...? It seems like your printer didn't return a correct json...

Ghostkeeper, 2017-03-27:
What's the version number of the firmware on your printer?

TinkerGnome, 2017-03-27:
Firmware version 3.6.1.20170308
It happened only one single time  - after the first print with this combination of Cura and printer.
I'll make a try with the 3.6.2 firmware tonight and report back.

TinkerGnome, 2017-03-27:
The exception is not reproducible (neither with 3.6.1 nor with 3.6.2), but perhaps it helps to "harden" the json decoder a bit...

Ghostkeeper, 2017-03-29:
The JSON decoder is built-in in Python.

Perhaps we should just catch this exception as a transmission error and only report it with a warning in the logs.

Ghostkeeper, 2017-03-29:
I just put some try:catch blocks around these: https://github.com/Ultimaker/Cura/commit/f62488fd110509e145af33281c0215750f7463c5

closed 2017-04-14 by TinkerGnome