Ultimaker/Cura #1590: Unable to add/modify post-processing script in 3.4 under Linux

buteomont, 2017-03-27:
I have downloaded the 3.4 Cura appimage file and it runs fine. However, I need to add my post-processing script to it, but Ubuntu won't let me modify the appimage file. 

How can I add my post-processing script to 3.4 running in Ubuntu Linux?

(P.S., I tried to look in the documentation for the answer but when I click Help -> Show Online Documentation nothing happens.  But that's a separate issue.)

nallath, 2017-03-27:
Currently that's the only place that the plugin can check for new scripts. 

I've created an issue on our tracker (CURA-3593)

randyxxl, 2017-03-27:
Issue is listed as CURA-3597

Ghostkeeper, 2017-03-27:
About the "Show online documentation", a few users reported that too but then it turned out that it added a tab to some Firefox window in the background or something, without them noticing.

ChrisTerBeke, 2018-01-22:
You can now place a custom post processing script in the user_data/scripts folder.

closed 2018-01-22 by ChrisTerBeke