Ultimaker/Cura #1592: infil density for extruder 2 (PVA) CURA 2.5 BETA does not work

BrugnettiKim, 2017-03-27:
Dear all,
I'm testing the new CURA software. I change the "infill density" value for extruder 2 (BB.04 PVA) but this changement does not affect the structure.

Thanks

Vandrasc, 2017-03-27:
Seems to work on my Win10. Is it printing with the second extruder? Extruder 2 has to be selected from Per model settings (on the left tools menu, under the mirror tool) then it should generate the correct infill density. Otherwise, it is still printing with the first extruder settings.

nallath, 2017-03-27:
Also; Please provide us with a project file. That way we can see what you are doing (wrong?)

BrugnettiKim, 2017-03-27:
It works but i cannot change the "infill density" for PVA material in the second extruder BB.04 .  
The value"infill density = 20" produce the same PVA structure than "infill density = 90". 
I work with windows 10.

See images : 

[image]
[image]

BrugnettiKim, 2017-03-27:
I found my error. I changed de bad value.

Thanks

BagelOrb, 2017-07-21:
This is one of the reasons we sometimes propose to make all settings inactive for an extruder which doesn't have any models loaded.
IT's kind of annoying to not be able to change settings without loading a model first, though.