Ultimaker/Cura #1593: machine_extruder_start_pos_*** does not affect the gcode. BUG ?

karabas, 2017-03-27:
cartesian printer, prusa like, dual carriage like.
If I setup "machine_extruder_start_pos_***" in def files no corresponding gcode is generated.
But "machine_extruder_end_pos_***" settings work and gcode is generated.

So I am forced to use start stop custom gcode settings, but faced with a problem of Z-lift. It applied AFTER my gcode.

BTW how to comment strings correctly in extruder.def files? I tried // and # and ;  Nothing works. Cura does not started.

fieldOfView, 2017-03-27:
comments are not allowed in json files

Ghostkeeper, 2017-03-27:
You can add fields like `"comment": "Cetero censeo Cartaginem delendam esse."`, just a JSON field that we don't use, because we ignore them.

As for the issue itself...

@BagelOrb will know this better, but I believe this is by design. We move to the end position because the printer expects the extruder to be there, but we don't need to move to the start position because the printer already puts the extruder at that position by switching. The start position is then only used for time estimates. These time estimates are displayed in Cura when slicing is done, but they are also important for making sure that the extruder heats up timely before the switch.

karabas, 2017-03-27:
If your printer has toolchanger or something like this, you need T0 end gcode to move drive carriage to park T0 tool. Then T1 command start gcode moves drive carriage to T1 tool and grabs it. All this in retract Zlift enable on tool change.

BagelOrb, 2017-03-27:
You can just put a gcode in which only moves the x:
G0 X0

You can also move up if you want, by using
G0 {zlift_setting_name}
in relative extrusion mode.

You can do the z lift in your start gcode.

karabas, 2017-03-28:
Yes. But it seems for me more intuitive and correct way to use machine_extruder_start/end settings as they are intented for this. 
BTW Is there any macro of current coordinate of the head? something like G1 Y{current_y_pos}?
Where can I find list of marcos to use in config.defs?

BagelOrb, 2017-03-28:
That's what I'm saying; you can add those gcode commands in the machine_extruder_start/end settings.

There is no macro for the current coordinate. You never need to send `G1 Y{current_pos}`, because the print head is already at that position. Instead you could just set `G1 X100` and not set the Y.

The 'macros' which you can use are just the settings defined in fdmprinter.def.json.

karabas, 2017-03-29:
Ok, I did it. Thanx

closed 2017-03-29 by karabas