Ultimaker/Cura #1594: Crash

CamoJackson, 2017-03-28:
Version: 2.5.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
TypeError: unable to convert a C++ 'QVector3D' instance to a Python object

aevyrie, 2017-03-28:
Same error. Been using without issue until now, tried enabling and disabling auto-slice. Cura is fine until I click on an object on the build plate (tried multiple .STL's), then I get this error, every time:

> Version: 2.5.0-BETA
> Platform: Windows-10-10.0.14393
> Qt: 5.7.1
> PyQt: 5.7.1
> 
> Exception:
> TypeError: unable to convert a C++ 'QVector3D' instance to a Python object

Strangely, I had just been working with the part and restarted because I changed the currency and a material price/weight.

nallath, 2017-03-28:
This has to do with the "center camera on select" issue. If you disable this it should be gone.

Bobb1, 2017-03-28:
I can confirm nallath's comment.  This only occurs when "center camera..." is checked.

Ghostkeeper, 2017-03-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/1566 .

closed 2017-03-29 by Ghostkeeper