Ultimaker/Cura #1595: Infill layer thickness not working 2.5 BETA

ChrisRiddell, 2017-03-28:
Hello,

If you set the Infill layer thickness to 0.2 on the normal profile nothing changes the print time stays the same if you look at the layer view mode and go through the layers its not doing 2 wall layers then an infill. Also comparing exported gcode there is a ;TYPE:FILL.... for every layer.

mqudsi, 2017-04-01:
It still works sometimes. Changing it to much larger/smaller values changes the print time.

ChrisRiddell, 2017-04-02:
Yeah seems to be very hit and miss especially on smaller STL's

Ghostkeeper, 2017-04-03:
If the infill volume is so small that it wouldn't fit multiple layers in vertically (or even no infill), the infill layer thickness obviously has no effect. It also has to make sure that these infill layers align with each other even if the bottom side of the infill volume gently slopes, and this makes fewer infill merges possible.

Ghostkeeper, 2017-04-03:
I can't seem to reproduce any problems with this.

mqudsi, 2017-04-03:
Thanks for the explanation, @Ghostkeeper. That might explain it, I usually use the "pyramid" infill which would definitely be harder to have multiples of the layer height.

ChrisRiddell, 2017-04-04:
@Ghostkeeper You are right.

ChrisRiddell, 2017-04-05:
@Ghostkeeper Just to confirm,

1) with a small model as long as the infill is not over 90% this works. (so far seems that way in my testing)

2) if you have a layer height of .2 if i put in .4 for the infill height the field goes yellow that's a multiple of .2 or once you go passed the 0.15 layer height does that layer height rule change?

Ghostkeeper, 2017-04-06:
1. The infill layer thickness has nothing to do with the actual infill density. It increases the line width of the infill at some layers but assumes that this is compensated by having no lines there at lower layers, so the density remains the same. It does not work however if everything is skin! And if infill density is 100% we set everything to become skin.

2. The infill layer thickness becomes yellow if you surpass 75% of the nozzle size, because we think it won't be able to extrude enough material any more then. The infill layer height is rounded to the nearest multiple of layer height.

closed 2017-06-27 by BagelOrb