Ultimaker/Cura #1596: Feature Request: Layer view mode, layer by layer button

ChrisRiddell, 2017-03-28:
Hello,

Something I want to request is some sort of up and down buttons next to the layer slider so its easier to go layer by layer in the layer view mode. The slider is nice but find it a little cumbersome if you want to go up or down one layer at a time. it would also be faster then it is to type and hit enter on the keyboard for each layer.

fieldOfView, 2017-03-28:
In Cura 2.5, the layer slider has become quite advanced. If you drag the top blue "nub" all the way down to layer 1, you get a single layer view. Now click the grey area inbetween the two grey "nubs" you should notice the layer number indicator aligning with the center of the grey area. 

With the center of the slider selected, you can use Ctrl + Up/Down to view one layer at a time. Add the Shift key, and the view will bump up/down by 10 layers. You can also enter a number in the layer number box to go straight to the specific layer. Finally you can also drag the grey area of the slider.

The same also works for arbitrary numbers of layers at the same time. If you manipulate the center of the slider, the top and bottom should stay the same number of layers apart.

ChrisRiddell, 2017-03-28:
Ahhh This is nice, These keyboard methods need to be documented some where. 

on MacOS it's the command and up/down arrow but it does give the system error sound :/

fieldOfView, 2017-03-28:
As a workaround, you can click in the layer number box and just use the up/down keys.

fieldOfView, 2017-03-29:
This issue can be closed (right?)

ChrisRiddell, 2017-03-29:
I'm going to say so.

closed 2017-03-29 by ChrisRiddell