Ultimaker/Cura #1597: Setting price for custom material changes price for all custom materials

JonasGe, 2017-03-28:


fieldOfView, 2017-03-28:
This is an effect of custom materials being a clone of another material. It is a design flaw of using "Duplicate" to create a New material.

JonasGe, 2017-03-28:
That's what I was thinking. Is there a way to add new material other than duplicating?

nallath, 2017-03-28:
Only if you change the files by hand. We don't have a good flow for creating a "new new" material.

Ghostkeeper, 2017-03-29:
The intent was to allow users to create new profiles for existing materials, such as to adjust the bed temperature. But in the GUI this was then communicated as if the user is creating actual new materials. There is no good workflow for this yet.

MarcusWolschon, 2017-04-21:
What exactly does a user have to do and where are the files on Linux, Mac and Windows?

Ghostkeeper, 2017-04-25:
On Windows the material files are in `%APPDATA%\Local\cura\materials`. On MacOS they are in `~/Library/Application Support/cura/materials`. On Linux they are in `~/.local/share/cura/materials`.

In the XML files, find the GUID field and modify it to have a different GUID. If it's just for yourself, you can change one of the numbers in the GUID to a different number. If done properly, you can execute `import uuid; print(uuid.uuid4())` in Python or just find an [online generator](https://guidgenerator.com/online-guid-generator.aspx) that generates random GUIDs.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-10:
This was fixed in 2.6 by allowing people to create actual new profiles rather than just duplicating them.