Ultimaker/Cura #1601: Not all layers shown when raft is used 2.5 BETA

PascalMa, 2017-03-29:
When using a raft, the model is not shown to the top in layer view.

In my case the layer height is 0.04mm.
Seems like the "Raft Air Gap" contributes to this, as it shifts the model up, but doesn't increase the index of the topmost selectable layer.
But even with Raft Air Gap = 0, there are still some layers missing and I haven't figured out yet wich parameter is not considered in the max. layer index calculation.

Ghostkeeper, 2017-03-29:
Indeed, this reproduces. Quite a serious bug! Thanks.

It seems to create a model of the same height regardless of raft height, including air gap and such.

Ghostkeeper, 2017-03-29:
For our reference: CURA-3615.

jackha, 2017-04-04:
Fixed

closed 2017-04-05 by nallath