Ultimaker/Cura #1603: Cura 2.5 crashing on start after duplicating material

Robinhuett, 2017-03-29:
I can't start Cura 2.5 BETA anymore. after I tried to duplicate a material it crashed and now it crashes on every startup with the folowing error:
```
Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\Models\InstanceContainersModel.py", line 46, in __init__
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\Models\InstanceContainersModel.py", line 61, in _update
TypeError: unorderable types: str() < NoneType()
```

This error is thrown in the "Loading Interface..." stage.
Before this 2.5 worked and 2.4 still works.

Ghostkeeper, 2017-03-30:
> 2.4 still works.
Huh. That suggests that it's something we changed in the display code rather than something that was saved to your profiles.

The line of code where the error occurs is this:

    self._instance_containers.sort(key = self._sortKey)

In `_sortKey` it gives a list of keys on which to sort:

    def _sortKey(self, item):
        result = []
        if self._section_property:
            result.append(item.getMetaDataEntry(self._section_property, ""))

        result.append(not item.isReadOnly())
        result.append(int(item.getMetaDataEntry("weight", 0)))
        result.append(item.getName())

So somewhere, this must be putting `None` in the `result` list. It can't be `_section_property` since we check for it to be `None` before putting it in the list. It can't be `isReadOnly` since it has type hints that it must return a boolean. It can't be `getMetaDataEntry("weight", 0)` since it has a default of 0 if there is no such entry. And it can't be `item.getName()`, because that also has a type hint that it should return `str`. Where did I make the mistake here?

Ghostkeeper, 2017-03-30:
I can't reproduce the bug, by the way.

LipuFei, 2017-04-04:
`getMetaDataEntry()` can be `None` if `_metadata` contains the key `self._section_property` with value `None`. `dict.get(key, default)` only returns the default when `key` cannot be found.

Suppose the typing hinting works, then the most suspicious one is
```
item.getMetaDataEntry(self._section_property, "")
```

nallath, 2017-07-27:
Closing it as we can't reproduce this.

closed 2017-07-27 by nallath