Ultimaker/Cura #1605: .obj "unable to load"

genic, 2017-03-29:
Hey, ... Been using Cura 15.04 so far.
Upgraded to 2.4 today. Got "unable to load ***.obj" on every file.
Dlownloaded 2.5 beta ... Got a crash report:

Version: 2.5.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 652, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 216, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Event.py", line 187, in call
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 124, in calculateBoundingBoxMesh
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 581, in getBoundingBox
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 648, in _calculateAABB
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 176, in getExtents
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 251, in getConvexHullVertices
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 243, in getConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 235, in _computeConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 365, in approximateConvexHull
  File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
  File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
scipy.spatial.qhull.QhullError: QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate

While executing:  | qhull i Qt
Options selected for Qhull 2015.2.r 2016/01/18:
  run-id 1394784687  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
  Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0

fieldOfView, 2017-03-29:
Could you perhaps post a link to one of those obj files so we can investigate?

genic, 2017-03-30:
Sure thing. Thank You for looking into it.
Actually all my .obj files that used to load and print with cura 15.04 are affected.
Here is one example.
https://www.dropbox.com/s/28jex98k5xuqjcq/tire.obj?dl=0
Thanks again.

genic, 2017-03-30:
And here is the crash report from the same model:

Version: 2.5.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 652, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 216, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Event.py", line 187, in call
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 124, in calculateBoundingBoxMesh
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 581, in getBoundingBox
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 648, in _calculateAABB
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 176, in getExtents
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 251, in getConvexHullVertices
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 243, in getConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 235, in _computeConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 365, in approximateConvexHull
  File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
  File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
scipy.spatial.qhull.QhullError: QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate

While executing:  | qhull i Qt
Options selected for Qhull 2015.2.r 2016/01/18:
  run-id 1727580094  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
  Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0

fieldOfView, 2017-03-30:
The problem is that Cura expects the object to be in mm, and this seems to be modeled in meters. That means that the coordinates in the objects are so small that some calculations break due to rounding issues.

If you scale up your object with a factor of 1000 before exporting, it should work. Having said that, Cura should be more resilient to this.

genic, 2017-03-30:
i understand. Thank You very much for checking.
My models are exported from Lightwave 3D. Cura used to understand the dimensions.
Sure would be nice if Cura would import the models correctly, eventually.
But, its a nice workaround for now. Thanks again.

genic, 2017-03-30:
Sorry dude, gave it a try but it actually does not work for me after all.
Scaling the model still does not import ... doesn't crash anymore though.
but still "failed to load ***.obj" :/

Ghostkeeper, 2017-03-30:
It shouldn't fail to load even if the coordinates have the wrong units. Indeed, Cura should be more resilient to this. I'm seeing the same error.

This file works.
[tire.stl.zip](https://github.com/Ultimaker/Cura/files/881545/tire.stl.zip)

Ghostkeeper, 2017-03-30:
The actual problem seems to be in QHull, the library we're using to compute numerous statistics about the model, such as the bounding box and the collision area. In other words, it's upstream.

fieldOfView, 2017-03-30:
But cura could/should still catch the exception and not crash because of the upstream issue

Ghostkeeper, 2017-03-30:
It's not crashing, at least not for me. It's producing a stack trace in the log and displaying a message on the screen that the file failed to load.

genic, 2017-03-30:
Right, for me ...
Cura 2.4 displayes "Failed to load xxx.obj" at the bottom but keeps running.
Cura 2.5 beta crashes down after giving me the reports that i posted.

Ghostkeeper, You have been also unable to open my .obj, right?
You converted the model to .stl to import it? did you also re-scale it?

fieldOfView, You were able to load the model after re-scaling? may i ask, what program did you use?

fieldOfView, 2017-03-30:
I have not tried to open the model in Cura. I just noticed that all the coordinates were very small and jumped to a conclusion (because I vaguely remember this being an issue before)

fieldOfView, 2017-03-30:
Pinging @sedwards2009, who did a lot of work on the convex hull stuff

genic, 2017-03-31:
i understand. Thank You guys.

Ghostkeeper, 2017-04-03:
> Ghostkeeper, You have been also unable to open my .obj, right?

Indeed, it gave an error message in Cura for me.

> You converted the model to .stl to import it? did you also re-scale it?

Yes, I scaled it x1000 and then saved as STL in CloudCompare.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-16:
Can confirm that this is fixed right now. We fixed the auto-scaling of small models for the 3.1 release. There were some problems in there.