Ultimaker/Cura #1610: CURA WONT OPEN

JasonBarretto, 2017-03-30:
Version: 2.3.1
Platform: Windows-10-10.0.14393
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

nallath, 2017-03-30:
Upgrade to 2.4? If it's a special version created by / for ROBO printers, contact them.

Ghostkeeper, 2017-03-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/1312, https://github.com/Ultimaker/Cura/issues/1201 and https://github.com/Ultimaker/Cura/issues/1107.

closed 2017-03-30 by Ghostkeeper