Ultimaker/Cura #1611: Cura 2.5 Beta - strange behavior with variable material_bed_temperature, not connected to printer?

tamarintech, 2017-03-30:
My start gcode reads:
`M140 S{material_bed_temperature}`

I receive the following warning:
`2017-03-30 13:00:12,111 - WARNING - UM.Settings.SettingDefinition._deserialize_dict [524]: Unrecognised property visible in setting material_bed_temperature`

But the output gcode renders as expected:
`M140 S80`

Not a big deal, but it seems like I'll receive this warning shortly after printing:
`2017-03-30 13:03:04,407 - ERROR - USBPrinting.USBPrinterOutputDevice._sendNextGcodeLine [569]: Unexpected error with printer connection: could not convert string to float:`

At that point, the Status window will show "Not connected to printer" while the printer *is printing*. There is no way to abort or disconnect when this happens outside of just restarting Cura. :(

If I remove all {material_bed_temperature} macros it works as expected.

Ghostkeeper, 2017-04-03:
Hmm, seems that it's using an outdated machine file. It's complaining that the property "visible" on the `material_bed_temperature` doesn't exist. That's correct, because we removed the "visible" property on all settings, in favour of making a setting invisible if it's not "active". Your machine definition still tries to override the "visible" property. In any case, it's a warning that you can ignore.

I looked through the definitions and I suspect that you might be using Renkforce RF100. I removed the property from the definition. Thanks for the notification.

tamarintech, 2017-04-03:
:+1:

I am using a brand new "Custom FDM Printer" created on startup. Maybe there's a leftover property in the initialization for that?

Ghostkeeper, 2017-04-03:
Could be if you went through using the Renkforce. The "visible" property is not present in the Custom FDM Printer definitions.

tamarintech, 2017-04-03:
I definitely did not do anything with a Renkforce.

Start Cura, Custom FDM. That's all.

Ghostkeeper, 2017-04-04:
I'm not reproducing it with these steps:

* Clear settings.
* Start Cura.
* Add Custom FDM Printer.
* In the machine settings action, set the start g-code to `M140 S{material_bed_temperature}`.
* Load a cube. Let it slice.

In the log there is no warning about the bed temperature.

Ghostkeeper, 2017-04-04:
I made a change to the Renkforce definition to remove the outdated property, but that won't have changed anything for you.

tamarintech, 2017-04-04:
:+1:

Thanks for sorting that out!
I'll try to recreate later, but maybe I have old Cura profiles it imported? But the only machines I've used with Cura have been Custom FDM. The renkforce thing is throwing me off. :)

Closing up!

closed 2017-04-04 by tamarintech