Ultimaker/Cura #1612: Cura 2.5 Beta, float error, not connected to printer?

tamarintech, 2017-03-30:
I'm getting a:
`2017-03-30 13:43:09,966 - ERROR - USBPrinting.USBPrinterOutputDevice._sendNextGcodeLine [569]: Unexpected error with printer connection: could not convert string to float:`
whenever I try to print.

What's the best way to find out whatline 569 might be? I tried saving the gcode, but line 569 reads `G1 X75.646 Y82.369 E12.2245` and seems just fine compared to 568, 567, etc.

It seems like this might be causing the print status window to constantly display "Not connected to printer" even though it's printing. The print is generally just fine after all of this, but I can't abort, pause or cancel it. :(

nallath, 2017-03-30:
I think the issue occurs within the python code, which is the line number it refers to.

tamarintech, 2017-03-30:
Oh! Duh. :) That makes a lot more sense....
Let's see...

`           if ("G0" in line or "G1" in line) and "Z" in line:
                z = float(re.search("Z([0-9\.]*)", line).group(1))
                if self._current_z != z:
                    self._current_z = z
`

I have lines like this in my startup gcode:
`G1 Z0  F1200`

Maybe the match is a bit greedy and trying to float("Z0  F1200") ?

tamarintech, 2017-03-30:
https://github.com/Ultimaker/Cura/pull/1615

tamarintech, 2017-04-02:
Oops.
Meant to close when I put the PR up. Closing now.

closed 2017-04-02 by tamarintech