Ultimaker/Cura #1616: Cura 2.5 Beta Fatal error when printing from gcode

andrewbecker, 2017-03-30:
When I load a gcode file into Cura 2.5 beta I am unable to print over network to UM3.  I get a fatal error.

Version: 2.5.0-BETA
Platform: Darwin-15.6.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 652, in event
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 216, in event
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Event.py", line 187, in call
  File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 116, in _writeToDevice
  File "/Users/adbecker/Applications/CuraBeta.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevice.py", line 624, in requestWrite
    if print_information.materialLengths[index] != 0:
IndexError: list index out of range

fieldOfView, 2017-03-31:
This was just fixed in master: https://github.com/Ultimaker/Cura/commit/c785e84b86effbeec0d0f46fcb6fc3803d579e2f
Perhaps this fix needs to be cherry-picked to 2.5?

LipuFei, 2017-03-31:
@fieldOfView we made a fix for this one on 2.5 too: https://github.com/Ultimaker/Cura/commit/c785e84b86effbeec0d0f46fcb6fc3803d579e2f

LipuFei, 2017-03-31:
Ah, it's the same commit.

fieldOfView, 2017-03-31:
Great! The commit only shows "master" in github, hence my remark.

closed 2017-03-31 by nallath

awhiemstra, 2017-04-03:
> Great! The commit only shows "master" in github, hence my remark.

Yeah, I consider that a github bug, it only shows one branch the commit is in.