Ultimaker/Cura #1617: Printer not connected, but it's printing

tamarintech, 2017-03-30:
I'm having trouble with the monitor showing that the printer is not connected, but I am able to print via USB. I see no obvious errors in the logs/console. The print completes successfully, but this isn't really ideal - I can't cancel or pause the print without resetting the printer (reconnecting, restarting Cura, etc).

As far as I could tell, the "printer connected" state should be triggered as long as M105 returns something sane. This is my M105 response when issued via separate terminal `ok T:30.5 /0.0 B:31.9 /0.0 B@:0 @:0`

Is that response not expected? Is there something else that would cause "Printer Disconnected" in the monitor?

fieldOfView, 2017-03-31:
As far as I can tell the M105 trick is only used to check the initial connection, not to check if the printer is still connected. Is there anything useful in the logs? It could very well be related to the negative Z value thing you fixed.

tamarintech, 2017-03-31:
I've tried it with the negative Z fix and even tried just removing all the negatives from my gcode. Tried adding a new FDM printer with a heated bed and default gcode start. No luck. :\

The logs show a successful autodetect on ttyACM0. I can select a model, pick "Print via USB" on the drop down, and click 'Print'. The print starts and completes successfully, but there is no status. Before the print, all of the bed prewarm, etc, controls are grayed out.

This happens in between slicing and printing, but nothing stands out to me as USB printing related.

`2017-03-31 03:54:17,602 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [447]: Slicing took 1.114391803741455 seconds
2017-03-31 03:54:17,604 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-03-31 03:54:17,606 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-03-31 03:54:17,608 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-03-31 03:54:17,608 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [545]: Backend quit with return code 0. Resetting process and socket.
2017-03-31 03:54:26,210 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles`

tamarintech, 2017-04-03:
The plot thickens...

Just for giggles, I decided that since I can print with Cura (still with no monitor) I should see if I could update firmware. I can. Immediately after the firmware update completes the monitor works.

Maybe there's a race condition or a reset missing in the USBPrinting plugin?

Ghostkeeper, 2017-04-03:
If it's a race condition or a reset, then restarting Cura would give a chance of making the monitor work again. But the monitor is pretty straightforward. It's just processing incoming messages and displaying the result of them on the screen. I wouldn't expect to find a race condition there, even if the messages come in out of order.

I'm placing my bets on using an old firmware version instead, or a corrupt Arduino installation on the other side.

I'm not reproducing the issue here.

tamarintech, 2017-04-03:
I have a LulzBot Mini and a reprap Wilson 2.
I believe the Wilson 2 is RC3. I would have to check the LulzBot version, but I am using whatever they roll with their latest Cura. The OS is Fedora 25 and I can reliably recreate the problem across two desktops and a number of cables.

I learned last night that if I repeatedly plug/unplug either one after starting Cura the monitor will eventually work and continue working until I restart Cura. O.o

Even if this is a whacky edge case maybe we could add some way to manually trigger a reconnect, or is there a way to do that now? It worked 100% of the time I used the firmware upgrade tool, but that sounds like a costly solution. :)

ivanporro, 2017-05-01:
I have the very same issue. 
Did some workarounds to get the printer connected but Cura's monitor still believe it is not.
I described my steps here:

https://github.com/Ultimaker/Cura/issues/1543#issuecomment-298429296

jackha, 2018-08-16:
afaik it works fine with the printers we tested it with. closing because of inactivity and lack of reproducibility

closed 2018-08-16 by jackha