Ultimaker/Cura #1618: Fatal exception: AttributeError: 'PrintInformation' object has no attribute '_material_amounts'

laurenweinstein1, 2017-03-31:
Version: 2.5.0-BETA
Platform: Linux-4.4.0-70-generic-x86_64-with-debian-stretch-sid
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/build/2.5/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 652, in event
  File "/build/2.5/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 216, in event
  File "/build/2.5/build/inst/lib/python3/dist-packages/UM/Event.py", line 187, in call
  File "/build/2.5/build/inst/lib/python3/dist-packages/UM/Signal.py", line 305, in __performEmit
  File "/build/2.5/build/inst/lib/python3/dist-packages/cura/PrintInformation.py", line 187, in _onMaterialMetaDataChanged
  File "/build/2.5/build/inst/lib/python3/dist-packages/cura/PrintInformation.py", line 134, in _calculateInformation
AttributeError: 'PrintInformation' object has no attribute '_material_amounts'

fieldOfView, 2017-03-31:
What were you doing when this happened? Can you reproduce it?

nallath, 2017-03-31:
I think this already has been fixed.

laurenweinstein1, 2017-03-31:
Triggered when reloading a saved project file. 100% reproducible.

fieldOfView, 2017-04-01:
Can you share the project file so we can check it is ondeed fixed?

laurenweinstein1, 2017-04-01:
I checked but unfortunately I no longer have that project file. I took a
different approach due to the recurring error at the time.

Ghostkeeper, 2017-04-03:
This was CURA-3617. It's been fixed by this commit: https://github.com/Ultimaker/Cura/commit/b3bc019cd0ea8b022dc020a30bf61538a07c4476

closed 2017-04-03 by Ghostkeeper