Ultimaker/Cura #1621: Unable to connect to Printer and Unable to start print CURA2.5beta

UltimateSports, 2017-03-31:
Print is fully prepared and sliced, 
printer shows up in configuration window, 

printer 'connects' and sits ready to print 

but UM3 will not stay connected have to resort to USB Removeable device when 'printer is busy' although there is nothing in progress.

Spirou42, 2017-03-31:
Have the same issue here. 
SetUp: Cura 2.5beta, UM3 (3.6.2.20170322), OSX. 
Network printing stops shortly after transferring file to printer.
Saved file to USB-Stick and tried USB printing.
Printer: "model to large for build volume" 
After selecting  [Ignore] the print runs fine.
It seams as if the Cura2.5b sliced files are detected as "too large to print"

nallath, 2017-04-03:
I think we have fixed this issue in the meantime.

Ghostkeeper, 2017-04-03:
Yes, it was fixed with this commit: https://github.com/Ultimaker/CuraEngine/commit/8106dfa88c701e86e79751e2b72ff5d17864a2da

closed 2017-04-03 by Ghostkeeper