Ultimaker/Cura #1623: Wrong file name when saving to SD card

uwe999, 2017-03-31:
I had a model open in cura 2.4 which I had previously saved the gcode to the SD card.
I then cleared the build plate,  selected a new stl in windows explorer, dragged and dropped it into cura.
When I clicked the "save to SD card" button it used the file name of the previous model.

Ghostkeeper, 2017-04-03:
I'm not able to reproduce this in 2.5 on Ubuntu. My steps to reproduce were:

* Insert an SD card.
* Open a file using the button on the top-right. Select Cube.stl.
* Save the file to the SD card. It saves as UM3_Cube.gcode on the SD card (expected behaviour).
* Clear the build plate using Ctrl+D.
* Drag Cylinder.stl from a folder into the Cura window.
* Save to SD card again. It now saves as UM3_Cylinder.gcode on the SD card (expected behaviour).

fieldOfView, 2017-04-03:
How about when you don't clear the buildplate by pressing Ctrl+D, but by selecting the Cube and removing it?

Ghostkeeper, 2017-04-03:
Same behaviour then.

uwe999, 2017-04-03:
seems like it only happens when you have a model open,
then drag another model onto the build plate and delete the first model.
Probably and edge case.

Ghostkeeper, 2017-04-04:
Ah, yes we only erase the name when the build plate is completely empty.

You can give your print job a name in the field on the bottom right. This name by default is a combination of the printer you're printing for and the file name of the first file you loaded. But you can set it differently if you like. And it'd be a bit annoying if you had to re-type your print job name every time you load a new file onto the build plate. So I'd say this is expected behaviour.

closed 2017-04-04 by Ghostkeeper