Ultimaker/Cura #1624: Enable prime tower error

Danielrug05, 2017-03-31:
Hi, 

I am doing dual printing with PVA and othe rmaterials, but when i see the layers mode in CURA, i can see an path that i later found was the one the printer would take to print the prime tower. Altough, i uncheck the "Enable prime tower" box, the path of the print keeps being the same as if i was printing with it checked. The print heads just go to the pint you will see below and stay there for 2 or 3 seconds, and then they continue printing. How can we fix that?
[image]

Ghostkeeper, 2017-04-03:
I'm not seeing this in the 2.5 beta.

Keep in mind that it does have to go to the switching bay in order to switch materials. On your screenshot, the switching bay seems to be in a fairly weird position (at least if it would be an UM3), but it looks like it's just the travel move to the switching bay, not a prime tower.

fieldOfView, 2017-04-03:
We don't know he's using an UM3 though.

Ghostkeeper, 2017-04-03:
No, I don't know. If it's not an UM3 it might not even be a fairly weird position.

Danielrug05, 2017-04-03:
Hi Guys, 

It is not a UM3, i have a custom fdmprinter and i took as reference the JSON file of the "Ultimaker_original_dual" to insert my own dual extrusion printer into the software, so i dont have a switching bay. Maybe its because of that? Do you know how can i solve it?

Ghostkeeper, 2017-04-04:
In the extruder definitions, set `machine_extruder_start_pos_abs` and `machine_extruder_end_pos_abs` to `false` and then set `machine_extruder_start_pos_x`, `machine_extruder_start_pos_y`, `machine_extruder_end_pos_x` and `machine_extruder_end_pos_y` to `0`.

This should remove any movement to a starting/ending position.

fieldOfView, 2017-04-04:
> This should remove any movement to a starting/ending position.

It could however cause the printhead to pause over the print while reheating the other extruder.

Danielrug05, 2017-04-04:
Hi, 

Thanks a lot Ghostkeeper! the problem has been solved. And the reheat is not a problem, just leave the standby temperature be the same as the printing temperature and the printing shouldn´t pause.

Thank you all

Danielrug05, 2017-04-04:
Hi Guys, me again. 
I thought that leaving the standby temperature and the printing temperature the same, will prevent the printhead to pause over the print, but it is still making that pause and it takes 10 seconds to move again. Maybe do you know why?

Thanks

karabas, 2017-04-05:
CURA2.5.beta.  I set all that temperatures to 210 and checked the gcode.
M109 are still there:
T0
G92 E0
G1 X0 F5002
M109 S210 

In cura 2.4 I simply hacked the exe file with hexeditor to replace M109 with M104

Danielrug05, 2017-04-05:
Hey Karabas, 

Did that solve the problem? I also set all the temperatures to the printing temperature.

Thanks

karabas, 2017-04-20:
I ended it hacking the binary curaengine.exe with hexeditor to change M109 то M104.

closed 2017-04-24 by Ghostkeeper