Ultimaker/Cura #1625: Cura 2.5 Dual Extrusion machine definition

stidf, 2017-03-31:
There is no way to define a dual headed custom extruder in Cura 2.5. I ave a custom dual headed printer that is a clone of a MBI rep 2x. The custom FDM printer profile doesn't give you the option of adding additional extruders in 2.5 and none of the other options appear to fit the bill. There are posts about a custom dual extrusion printer profile in 2.3 from the support forum.
https://ultimaker.com/en/community/22409-setup-custom-dual-extrusion-machine-in-cura-23-beta

So what happened to these features and will these features be coming back?

fieldOfView, 2017-04-01:
I'm working on support for multi-extrusion in the Machine Settings dialog for Cura 2.6:
https://github.com/Ultimaker/Cura/pull/1521

I wrote the Machine Settings plugin as a hobby project while working for Ultimaker, and have been continuing to improve it since I left Ultimaker. Because it doesn't directly affect the "happy path" for Ultimaker printers (for which there is no need to use the Machine Settings), it does not get much attention from the dev team at Ultimaker, but my pull requests for additional functionality are mostly met with enthusiasm.

GeoMagician, 2017-04-23:
Good to hear that this will go on your list. Thank you for dedicating your time to this issue to make us happy ;-)

BagelOrb, 2017-07-23:
fixed

closed 2017-07-23 by BagelOrb