Ultimaker/Cura #1626: cura keeps crashing on change or save profile

UltiArjan, 2017-04-01:
I've seen too much kittens today :)
cura keeps crashing when want to save a profile or change to a different profile

hope this will help;

Version: 2.5.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\ContainerManager.py", line 491, in createQualityChanges
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\QualityManager.py", line 270, in getParentMachineDefinition
KeyError: '1'

UltiArjan, 2017-04-01:
I noticed that even though i made changes to the pofile (see star) the option to update the profile is not available.

[image]

nallath, 2017-04-03:
Oh wow, that's a nice one. Could you share the project files?

UltiArjan, 2017-04-03:
Unfortunately I don't have a project file... but it now keeps crashing even without a loaded model.
I guess it has something to do with the machine definitions? though it was fine in 2.4

here's another one;

Version: 2.5.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 811, in setActiveQuality
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 916, in _determineQualityAndQualityChangesForQualityChanges
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\QualityManager.py", line 270, in getParentMachineDefinition
KeyError: '1'

to be complete, the machine and extruder definitions;
[mark2.zip](https://github.com/Ultimaker/Cura/files/891017/mark2.zip)

UltiArjan, 2017-04-03:
Just found out it always crashes when changing to or between a "custom" (saved) profile, but I can switch to one of the standard profiles and between the standard profiles.  Any change to a saved profile (that worked fine in 2.4) will crash cura.

dintid, 2018-02-13:
This bug just started happening in the new Cura 3.2.

Ghostkeeper, 2018-03-07:
What are the reproduce steps now then, Dintid? I can save and change profiles fine but nothing crashes.

dintid, 2018-03-07:
You fixed it in the quick 3.2.1 update. I "fixed" it myself before the quick update by manually removing any profiles and creating new ones.

Ghostkeeper, 2018-03-07:
Okay, thanks for re-testing then :smiley:

closed 2018-03-07 by Ghostkeeper

dintid, 2018-03-07:
Sorry I didn't follow up myself. But thank you for doing so 👍