Ultimaker/Cura #1630: Fields insist on losing focus when replacing existing text

mqudsi, 2017-04-01:
Currently testing under Cura 2.5 beta on Windows 10, but this is a very old issue:

If you click in a populated settings field and hit ctrl+a to select all existing text/value, then begin typing a new value, nine times out of ten, Cura will lose/change focus away from the text field either before you begin typing or in the middle of typing resulting in an incorrect value in that field.

This is especially irksome if auto-calculate is enabled as it means re-calculating several times with the wrong value (so useless results).

Ghostkeeper, 2017-04-03:
It could be happening because we register Ctrl+A as a shortcut to select all models, and backspace is a shortcut to delete the selected model.

I'm not reproducing this on Ubuntu however. Perhaps it's Windows-only?

mqudsi, 2017-04-03:
I don't think it's a global shortcut thing since no model is selected or deleted when I use these shortcuts while focused on a field. It must be something else, it's almost like Cura moves on to calculating as soon as the field changes rather than when it loses focus and *that* causes the loss of focus?

mqudsi, 2017-04-03:
OK, I've spent some time just now playing with this and I think I know what is happening. It's not that the input field loses focus, it's that changes to the "layer height" field are propagated to dependent fields in real-time rather than after the layer height field loses focus. This recalculation happens in the main thread, causing it to block (see #1631). It _appears_ to have lost focus since the blinking cursor disappears during this time. I know absolutely nothing about Python UI toolkits, but it seems that when the main thread is blocked, keystrokes posted to the main thread are lost (not queued) and so the input is lost in the same way it would be if focus disappeared.

mqudsi, 2017-04-03:
There may be more than one bug at play here. My last comment is what I run into most often, but there may be a larger underlying issue where actions are taken on the value_changed event (or whatever it's called in Python) for textboxes instead of focus_lost. The screencast [here](https://www.youtube.com/watch?v=IjLH_C_gSQk&feature=youtu.be) demonstrates the issue, that "12%" scaling was in the middle of my typing "120" but the focus was lost after the first two characters and 12% was applied instead.

Ghostkeeper, 2017-04-04:
Layer height changing should not make you lose focus on a setting textbox. Only a change to the profiles should be doing that. If that happens in other situations that is a bug.

I still can't reproduce it though. Perhaps some developer on a different platform could test this.

awhiemstra, 2017-04-04:
Calculations happening on value change is expected behaviour. We originally only used the "edit finished" signal but then got complaints that nothing happened if you just typed in some value. However, changing a value should not take more than ~50ms. It seems to have regressed badly recently, because I can see it takes way more than that now.

Ghostkeeper, 2017-04-05:
awhiemstra found that most of the delay results from checking the settings for errors. He's going to try introducing a delay in its calculation so that the settings still update real-time, but the red and yellow error colours only update after you're done typing.

awhiemstra, 2017-04-05:
It's not that actually - we do a check to see if anything has an error somewhere. That check was being done as part of the value changed handler, but was rather heavy code. I have now moved that checking code to a separate timer-based trigger, which means multiple change events will be combined and the error checking code will not run as part of the value change handler.

ianpaschal, 2018-08-07:
This no longer reproduces. Must have been fixed in the last year or so...

closed 2018-08-07 by ianpaschal