Ultimaker/Cura #1631: Reduce blocking calculations done in main UI thread

mqudsi, 2017-04-01:
This is a bit of a vague issue, but an effort is needed to reduce/eliminate the performing of blocking operations (especially calculations) in the main UI thread.

Currently (as of 2.5 beta), any number of changes to numeric settings values cause a number of calculations and consequences to be performed that take a not insignificant amount of time to run, tying up the main thread pump and freezing the UI for a period of time that ranges from 100ms to several seconds. It contributes significantly to making for a rather unpleasant user experience.

There are two bugs that I see: first is that the calculation of changes to other, dependent settings fields should be lightning fast (i.e. if field y is set to a multiple of field x, calculating the value of field y after a change to field x should take only a few CPU cycles at most). I don't know if this is because the lookup of dependent fields takes so long (in which case some sort of KV datastructure should be used, or a reference to dependent fields kept in memory somewhere) or (as I suspect) other non-UI-related actions are being performed as a result of the value change before the UI-blocking change-of-dependent-fields is performed (e.g. perhaps some slicing calculations are performed based on the change of x before the call to recalculate y based on the same change and update its value in the UI is performed). This would be an example of code that belongs in the UI thread taking too long to perform, hanging the application.

The second is that code that has nothing to do with the UI is being performed in the UI thread, this includes any form of slicing-related actions, render-related operations, etc. basically anything that does not need to be done in the UI thread being done there for some reason, causing the application to hang once more.

I realize these are vague issues that represent a not-insignificant undertaking for the team. However, I personally feel that these unfairly contribute to an unpleasant user experience that is not in keeping with the general quality of the Cura codebase and its awesomeness as a slicer. The solution will probably involve at least some minor changes to the workflow and will be a real PITA, but this is probably an issue that will only become more damaging and even more imperative to address the longer we wait, as it is only natural for the number of calculations and the activities/function calls invoked as a result of UI operations to grow as the codebase matures and further functionality is added.

Please note that I personally have an insanely fast workstation (32GiB of RAM, Xeon 1505M v6, NVMe SSD) and yet Cura still freezes the UI too often and for too long - it's an indication that far more work is being done in a synchronous manner than should be the case. Other users on more typically configured workstations will likely have an even less pleasant experience of it all.

I love Cura and this is in no way meant to disparage the efforts of the development team thus far. It's an awesome slicer and a beautiful example of open source done right. I just feel addressing this will contribute exponentially to making this great software even better. My apologies if I've been harsh or blunt above, no offense was intended! If any further clarification is needed or high-level suggestions for the workflow/architecture are needed, I am more than happy to help. Unfortunately, my unfamiliarity with the codebase makes me the last person that should attempt this undertaking, as it probably requires intimate knowledge of the flow of the software.

nallath, 2017-04-03:
We're well aware of issues in this regard. We heavily rely on signals & slots to update the UI. This can cause what we call "event storms" when a single changes trigger a lot of other changes, which in turn trigger even more changes. 

We've taken a number of steps to greatly reduce the odds of this happening, but as you correctly see, we're not there yet.

I'm not sure how familliar you are with Python, but as you may or may not know, Python has a GIL (global interpeter lock). So even if we use threads, which we do for a number of operations, it's not really multi threaded.

All that being said, feel free to keep us sharp. If you ever feel like helping out, drop a line. We're more than willing to help people to get started with Cura.

awhiemstra, 2017-04-03:
Many problems are related to Python and its GIL. While in theory, Python's GIL should only block byte-code level operations, the real-world results we get show that often the GIL gets locked to a single thread while that thread is performing calculations, giving no time to the main thread to actually update the screen. This means that moving things to a separate Python thread does not actually result in things being performed properly in the background, which leads to unresponsive UI. This also means that there is actually less incentive to do some things threaded, since it does not necessarily improve things.

We have come up with a few workarounds for this problem, one of them being to effectively introduce a lot of time.sleep(0) calls in the threaded code. time.sleep(0) is a special case of sleep that effectively forces Python to reconsider its thread scheduling and giving the main thread some time to perform updates. The other solution is to move more and more code into native Python extensions. This is why we introduced libSavitar for example, and why libArcus was reworked as a C++ module with Python bindings. The third thing we do is extensively profile what is happening and try to reduce the bottlenecks. The problem with that being that that often leads to less clear code, since we need to perform additional caching and similar things.

I am curious however, which exact things you are running into that take more than a few hundred milliseconds. As far as I know, there should be no setting calculation that takes that long. A few hundred ms may still be too much, but would be less of a problem than a few seconds.

LipuFei, 2017-04-03:
According to my old experience, I think [twisted](https://twistedmatrix.com/trac/) will solve a lot of those problems. :)
But I'm not sure whether it supports python3 well enough.

awhiemstra, 2017-04-03:
According to that site, "Twisted is an event-driven networking engine". Since we do not really need it for networking, I do not think it is going to solve that much for us. Also, the image on the front page (https://twistedmatrix.com/trac/raw-attachment/wiki/AttachmentBucket/oreilly_book_cover.jpg) reminds me a bit too much of spaghetti. :p

sedwards2009, 2017-04-03:
Qt will do event driven networking too, and is a better technology fit if you really need it.

mqudsi, 2017-04-03:
@awhiemstra:
> I am curious however, which exact things you are running into that take more than a few hundred milliseconds. As far as I know, there should be no setting calculation that takes that long. A few hundred ms may still be too much, but would be less of a problem than a few seconds.

I was playing around with this for #1630 and tracked down the bad experience to changes to the layer field. Changes there seem to trigger a bunch of calculations even before the field has lost focus, hanging the main thread for a few hundred ms at a time. I know the layer height is the most critical of the profile values and causes recalculation of many fields, but it really should be a few hundred cpu cycles of recalculations if everything is already in memory and correctly indexed. 

There are some other fields that exhibit similar behavior, but it's probably easiest to just play around with the layer height and track down the bottleneck from there; I presume it'll be a shared workflow with the other fields as well.

Ghostkeeper, 2017-04-04:
> but it really should be a few hundred cpu cycles of recalculations if everything is already in memory and correctly indexed

You're underestimating the overhead of Python and the complexity of Cura there. It has a list of the settings that are dependent on layer height. For each of these it has a 'value' that is really a string. It requests the current value of the setting from the top of the profile stack, first trying if there is a value for it in the user changes, then any user-defined quality profiles, the quality profile, the material profile, the extruder variant, the extruder definition, the global user changes, the global user profile, the global quality profile, the machine variant, and then finally in the machine definition. In the worst (and most common) case, the setting is finally found in the machine definition. But there it is still a string of Python code, which has to be compiled and then evaluated. And afterwards, all its other dependent settings have to be updated as well.

I think we don't cache the compiled versions of each inheritance formula yet. Perhaps that's an optimisation we could consider.

awhiemstra, 2017-04-04:
I created CURA-3653 in our internal tracker to track the issue about setting fields being unresponsive.

I also already have a fix for it, there was some code executed when changing from a valid value to an error value that would block the main thread. Moving this to a timer so it gets handled later greatly reduces the delays I am seeing when editing layer height. See the linked PR for details.

jackha, 2018-07-18:
Fixed from Cura 2.5 on.

closed 2018-07-18 by jackha