Ultimaker/Cura #1633: Decouple profiles for multi-extruders

mqudsi, 2017-04-01:
Cura seems to have a bit of an identity crisis when it comes to dealing with profiles in a multi-extruder setup. In some places in the UI, a profile is treated as a function of a single extruder:

[image]

[image]

In the above images, the profile is treated as an independent function of the extruder, i.e. there is a profile for each extruder. Which makes sense as the settings for material x in extruder 1 have absolutely nothing to do with the settings for material y in extruder 2 and they may be mixed-and-matched in any configuration.

However, while the settings for each extruder are saved in separate files (`$machine_extruder_{left,right}_$profile.inst.cfg}`), a profile seems to actually be a function of (current extruder 1 settings, current extruder 2 settings) and are saved and loaded that way. There doesn't seem to be a way (or, at least, an easy way) to load only the settings for extruder 1 from one profile and the settings for extruder 2 from another.

I would humbly suggest that the intent of profiles as implemented in Cura is that each extruder is to have its own, separate profile and that the inactive/unselected extruder should not be included in a profile for the active extruder nor affected by a load of a profile for the selected extruder.

fieldOfView, 2017-04-02:
There's a bug with the extruder menus - which I frankly don't remember if it ever got properly reported.

The menus will always change the setting of the "currently active" extruder, no matter which extruder menu you use. If you have the Extruder 1 tab selected in the settings sidebar, and then use the application menu to change the material, print core or profile of Extruder 2, Cura will actually make that change in Extruder 1. This is not correct.

As a workaround for now, please try using the sidebar instead of the menu.

There is also a "thing" that the two print cores can not print with different layer heights. So Cura tries to keep the two extruders at the same basic quality. Ie when you select a high quality profile in one extruder, the other will also switch to a high quality profile. But the two extruders can have _different_ high quality profiles. Note that I am not using capitals when saying "high quality"; it can be any profile derived of the "High Quality" profile.

Ghostkeeper, 2017-04-03:
The issue that mqudsi describes is one of the complex under-water functionality things that Cura does.

Basically, there are some settings, such as layer height or platform adhesion, that must be the same for all extruders. Cura can't handle different layer heights per extruder, and platform adhesion merges together when placing objects close to each other so it's impossible to separate them per extruder.

That's why we defined a "quality type" metadata property in every quality profile. Cura makes sure that all extruders have the same quality type at all time. For instance, UM3-0.4mm-PLA-Draft has a quality type "draft", and while that profile is loaded in the left extruder, a profile with "draft" type has to be loaded in the right extruder as well. If you have ABS in the right extruder, it'll find UM3-0.4mm-ABS-Draft. Both of these quality profiles have the same display name: "Draft" and this is why most people never notice that it's actually two profiles for two extruders.

When you create your own profile, it is linked to the same quality type as the profile you had selected when you created it.

In short, yes, you can select different profiles independently from each other for each extruder, but this is severely limited by all the settings that must be equal for all extruders, and we have a system in place that limits it.

fieldOfView, 2017-04-03:
> When you create your own profile, it is linked to the same quality type as the profile you had selected when you created it.

But unfortunately to make matters more confusing, it is possible to make a profile based on eg Draft that has a layer height of 0.06, which screws up the above reasoning.

fieldOfView, 2017-04-03:
@Ghostkeeper:  the menu-bug I describe in my comment, is that a known issue? Or shall I log that in a separate issue?

Ghostkeeper, 2017-04-03:
> But unfortunately to make matters more confusing, it is possible to make a profile based on eg Draft that has a layer height of 0.06, which screws up the above reasoning.

When the user makes the change it is put in the global profile.

>  the menu-bug I describe in my comment, is that a known issue? Or shall I log that in a separate issue?

I can't find an open ticket in Jira about it. I didn't know about it either. Could you make an issue?

fieldOfView, 2017-04-03:
Done: https://github.com/Ultimaker/Cura/issues/1641

mqudsi, 2017-04-03:
It seems there are a lot of complicated issues here, but I'll just re-phrase the original issue from a UX perspective.

From the sidebar, select a profile for extruder 1. Then switch to extruder 2 and select a different profile. Switch back to extruder 1 (still in the sidebar). The profile that was loaded for extruder 2 is now selected for extruder 1 (or, at least, shown as selected).

Ghostkeeper, 2017-04-04:
If you select a different default profile in the second extruder, it'll make sure that a profile of the same quality type is also selected in the first extruder. So if you select "High" in the second extruder, it'll also select "High" in the first extruder. For the default profiles these two happen to have the same name.

mqudsi, 2017-04-04:
@Ghostkeeper how is this determined for custom profiles? e.g. I have a custom profile called "ABS 0.2" and a custom profile called "PETG 0.2". Setting one extruder to one sets the other to the same.

fieldOfView, 2017-04-04:
You originally created these profiles by making changes to a base profile with a quality (eg Draft). So for Cura these custom profiles are seen as a draft profile.

mqudsi, 2017-04-19:
There's one other problem: because of how the profiles are set up, they are associated with extruder bank, meaning if I am printing PETG and ABS one day and configure the ABS extruder perfectly and I want to print ABS and ABS the next, I cannot use the ABS profile I created and must manually copy over all data from the second extruder to the first.

Unless I'm gravely mistaken, of course.

Ghostkeeper, 2017-04-24:
If you create a profile in the UI, the profile you create should be usable after you switch materials. So for these steps:
* Switch to PETG/ABS
* Change infill density to 42%.
* Save the profile, naming it "Infill 42".
* Switch the left extruder to ABS, so you're now ABS/ABS.

The expected result is this:

* "Infill 42" is still available as a profile that can be selected.
* "Infill 42" is still selected as the active profile.
* The infill is set to 42%. The rest of the settings are the default ABS settings (instead of previously the PETG settings).

jackha, 2018-07-18:
deferred because of inactivity

mqudsi, 2018-07-18:
I know this was just marked as deferred, but that spurred me to review the situation. I do not believe @Ghostkeeper's approach covers this problem.

The problem isn't with things like infill percentage but rather extruder-specific settings like temperature and movement rates combined with profile-level settings like layer height. I have a profile perfectly tweaked for PETG/PLA at 0.1mm, and want to print something in PETG/PETG at the same layer height. There's no way to switch PLA to PETG but get the per-extruder settings for extruder 1 to the second tab for extruder 2.

Also just as a matter of principle/abstraction, there are settings that are print-specific settings (let's call that PSETTINGS) and settings that are specific to a tuple `(PSETTINGS, EXTRUDER, MATERIAL)`. The second is by far more important to save than the first, but currently they are lumped together into one entity called a profile.

Tommmmmmmmmmmmm, 2018-07-21:
Yes, this issue has really been bugging me! I am excited to see @mqudsi commented 2 days ago though, because I think we need to discuss this further!

I think that as a quick fix giving us the ability to sync ALL settings between extruders would give us a quick workaround and save us manually copying settings over. In future I might try tuning settings whilst my printer is set as a single extruder and then apply that profile when in multi-extrusion and see if it gets applied to both extruders. Because once my printer is tuned I really hate having to manually copy settings over to another extruder when I want to use my material specific profile on another extruder.

Ghostkeeper, 2018-08-31:
> I have a profile perfectly tweaked for PETG/PLA at 0.1mm, and want to print something in PETG/PETG at the same layer height. There's no way to switch PLA to PETG but get the per-extruder settings for extruder 1 to the second tab for extruder 2.

There are two main use cases for profiles as far as I'm concerned. One is to enable your favourite features, such as ironing and concentric first layer, which are not really sensitive to the material. The other is to tweak the print quality for a specific (new) material and quality level. For the second use case, it's not useful to see the profiles made for different materials. For the first use case, it is useful. I don't think it would be user friendly to allow the user to switch between these two behaviours though. The settings system is complicated enough as it is.

> I think that as a quick fix giving us the ability to sync ALL settings between extruders would give us a quick workaround and save us manually copying settings over.

This is implemented now. You can right-click on the settings list and then copy all settings to all extruders.