Ultimaker/Cura #1635: Profile override confirmation dialog does not reflect correct values

mqudsi, 2017-04-01:
[image]

In the image above, the first column is incorrect as the saved value in the profile is actually 0.97 but in 2.5 beta it says that values differ though it shows them to be the same.

fieldOfView, 2017-04-02:
I think the default is a "calculated" value, the customized value is an "absolute" value. Ideally this difference would be visible.

mqudsi, 2017-04-02:
I'm not positive what you mean, but I'm pretty sure that 0.36 is neither the saved profile value nor what would have been calculated by the saved profile?

fieldOfView, 2017-04-02:
Ah, I thought you had manually entered 0.36, and were wondering why Cura was showing that it was a changed value. But it turns out I am having trouble reading today.

Ghostkeeper, 2017-04-03:
I filled in 0.97 and it showed both as 0.97 for me, even the default.

ianpaschal, 2018-02-27:
This is fixed in 3.2.1.

closed 2018-02-27 by ianpaschal