Ultimaker/Cura #1637: Ctrl key resets scaling options

Thor368, 2017-04-02:
I found this bug when i copy/pasted a scaling factor to cura:

If you tap the ctrl key the internal state of the option box "Uniform Scaling" is overriden with the "true" state. If the option box was "unchecked" before (therefore "Uniform Scaling" disabled) this leads to inconsistencies as the actual state is not reflected by the option box itself (it still shows "unchecked" state).

This is caused by the keyboard handler setting the internal state with absolut values. This overrides the user selected value in the scaling submenu without updating the option box.

I think this should be fixed by using xor operations on every ctrl key event. This toggles the state and therefore restores the state to its original value after the user releases the ctrl key.

Affected Cura Version 2.4.0

Ghostkeeper, 2017-04-03:
Also reproduced in Cura 2.5.

ianpaschal, 2018-02-27:
Fixed in 3.2.1.

closed 2018-02-27 by ianpaschal