Ultimaker/Cura #1638: Layer view getting slow with complex objects (until hang-up)

Dim3nsioneer, 2017-04-03:
The layer view in Cura 2.5 beta is generally great. However, with more complex object such as the UM3 Gyro, it slows down the GUI and makes it unresponsive. On one i7 PC with Win10 and an older graphic card (Radeon 6450) it managed to hang up the GUI entirely when having the Gyro scaled to 100%. It still worked (slowly) with the Gyro at 70% scale.

fieldOfView, 2017-04-03:
Could you test if "legacy mode" works better on that machine? See the General pane of the preferences.

awhiemstra, 2017-04-03:
We found during testing that graphics cards with limited shader performance (like Intel IGPs) tend to perform rather poorly when dealing with the new Layer view, especially when zoomed in. I would recommend (as @fieldOfView also suggested) to switch to the old layer view on such machines.

For some technical background: The new layer view does a lot of processing on the GPU. We use geometry shaders to perform the conversion from 2D polygon data to extruded tubes. This generally works well, but some graphics chips have problems with it, even if they support geometry shaders.

Dim3nsioneer, 2017-04-03:
The legacy view indeed works much better on the old machine. Is there any good way to tell users when to use that legacy view? Maybe the release notes?

closed 2017-04-03 by Dim3nsioneer