Ultimaker/Cura #1639: where can i get help with developing a plugin for Cura 2.5 beta?

zoff99, 2017-04-03:
https://github.com/zoff99/ElectricPrintCostCalculator

the plugin already works, but i want to access a few values in the plugin.
* printing time
* printing cost (only PLA cost now in 2.5)
* material weight

so i can use those in calculating the realistic cost (electricity and more ...)

fieldOfView, 2017-04-03:
See https://github.com/Ultimaker/Cura/blob/master/resources/qml/JobSpecs.qml#L202 about how to get the lengths, weights and costs based on what is entered into the Materials pane of the preferences.

zoff99, 2017-04-03:
i would need the calculated data. not the static data,

these 2 values:

[image]

so i do NOT need to enter them manually in the input fields here:

[image]

fieldOfView, 2017-04-03:
You can get them from PrintInformation.currentPrintTime and PrintInformation.materialCosts respectively. Note that both of these return an array as they can include information from multiple extruders (which you will have to add up for this plugin).

zoff99, 2017-04-03:
any chance of putting a feature like this into Cura itself?

fieldOfView, 2017-04-03:
That's not up to me to decide, but I don't think the Ultimaker devs will get much time to spend on this so it would have to come from a community member like you and I. 

It should work fairly well as a plugin downloadable, right?

zoff99, 2017-04-03:
is there a central plugin list or registry where i can link mine to?

nallath, 2017-04-03:
We're probably not going to build it, but by all means develop it! At the moment we are actively looking into making some sort of plugin repository so it will be easier to discover & share plugins.

Also note that it doesn't have to be a plugin. If it makes way more sense to put it in the "main" code, that's acceptable as well. Do note that we will review & test changes outside of plugins a whole lot better, so it could lead to a delayed acceptance.

zoff99, 2017-04-03:
wow thanks a lot. this is already working now.
only thing i need is to know when slicing/printtime changes? what signal do i need there? how do i catch it?

zoff99, 2017-04-03:
and also how can i get the currency symbol ($,€) that is set in the settings?

fieldOfView, 2017-04-03:
The values from PrintInformation.currentPrintTime and PrintInformation.materialCosts should automagically update. The currency symbol can be accessed with UM.Preferences.getValue("cura/currency")

zoff99, 2017-04-03:
yes they do, but my plugin does not update :-)
not sure where i can just say do it again ...

fieldOfView, 2017-04-03:
The best/easiest way is to get the values in QML instead of in python. You can remove most of the logic in the python part of your plugin (because it is already made available to QML from another part of Cura).

zoff99, 2017-04-03:
can't get the values to update. it works only on the first call.
don't know what i am doing wrong :-(

fieldOfView, 2017-04-03:
You can't get the values to update _because you are trying to do this in Python_, instead of in QML. You have based your plugin on a rather old plugin by @nallath, and by now there are better ways of doing what you want to do.

It is possible to make it work in Python (ofcourse), but it can be done much more elegantly in QML now. Look how JobSpecs.qml handles the values.

zoff99, 2017-05-29:
https://github.com/zoff99/ElectricPrintCostCalculator

so the latest version now works pretty nice.
can you guys link it to

https://github.com/Ultimaker/Cura#third-party-plugins

please?

nallath, 2017-05-29:
71bb3c3a22b9b21d761af15f0c3f72a4aed553dd There ya go.

For the next time, you can also make a pull request on our readme right away.

Ghostkeeper, 2017-05-29:
Without testing it myself, I wonder. How significant are the electricity costs compared to the filament, typically?

BagelOrb, 2017-07-21:
I suppose this question is answered.

closed 2017-07-21 by BagelOrb