Ultimaker/Cura #1641: Extruder application menu always acts on currently active extruder

fieldOfView, 2017-04-03:
Description:
The application menus for an extruder will always change the setting of the "currently active" extruder, no matter which extruder menu you use. If you have the Extruder 1 tab selected in the settings sidebar, and then use the application menu to change the material, print core or profile of Extruder 2, Cura will actually make that change in Extruder 1. This is not correct.

**CURA Release:**
2.3 - master

**Bug Type:**
New Bug

**Platform:**
any

**Graphics Card:**
any

**Operating Sytem:**
any

**Machine:**
Jedi, any multi-extruder

**Community Issue:**
https://github.com/Ultimaker/Cura/issues/1641

**Preconditions:**
A multiextrusion printer.

**Reproduce steps:**
* Add a new UM3 printer to Cura, do not connect it
* Note that both extruders have PLA for material and that Extruder **1** is the selected extruder tab in the sidebar.
* From the Settings application menu, choose Extruder **2** -> Material -> ABS

**Expected result:**
The material of Extruder 2 (2nd tab) changes

**Actual result:**
The material of Extruder 1 (1st tab) changes instead

**Report types:**
1) Coding error

**Screenshot:**
n/a

**Model:**
n/a

**Logs:**
n/a

"It's an old template, but it checks out."

Ghostkeeper, 2017-04-03:
This is my current template by the way:

    *Description:* 
    *Version:* 
    *Operating System:* 
    *Machine:* 
    *Reproducibility:* 
    *Reproduce steps:*
    # 

    *Actual result:*
    * 

    *Expected result:*
    * 

    *Keywords:*

Ghostkeeper, 2017-04-03:
Reproduces on 2.5.

Ghostkeeper, 2017-07-03:
This issue was closed by our product manager because it has been inactive for 3 months.

fieldOfView, 2017-07-03:
It is still a bug though.

Closing feature requests due to inactivity is one thing. Closing reproducible bugs is quite another.

Ghostkeeper, 2017-07-07:
I typically leave the issue open here on Github, but I do communicate the Project Manager's decision here.

The argument for closing bugs is that they sometimes expire (i.e. are no longer bugs) because of changes in the code, and that if nobody comments on the issue then it must not be that important.

The first argument doesn't really hold for closing features though. Feature requests don't really expire.

The second argument doesn't really hold if the decision to close them is based on the internal Jira ticket rather than the Github issue, which is the case for us. We've had cases where the Github issue was very busy but we don't copy all of the updates and comments to Jira and the Jira ticket was closed anyway.

There's also the third argument that keeping the list of issues short makes it easier to order them to their relative importance. The solution to that would be to have a separate category of issues in Jira of "probably never"-like issues. But in the end they are deleted from that category as well so that's not used very consistently either.

PR #2224 (2017-08-23): Fix multiextruder menus
This PR fixes the application menus for materials and variants for multiextrusion printers. All extruder menus would always act on the currently active extruder only, instead of acting on the extruder suggested by the menu. 

**Reproduce steps:**
* Add a new UM3 printer to Cura, do not connect it
* Note that both extruders have PLA for material and that Extruder **1** is the selected extruder tab in the sidebar.
* From the Settings application menu, choose Extruder **2** -> Material -> ABS

**Expected result:**
The material of Extruder 2 (2nd tab) changes

**Actual resu [... cut]
closed 2017-08-23 by alekseisasin