Ultimaker/Cura #1643: If possible, asynchronously load models dragged to Cura

mqudsi, 2017-04-03:
Dragging-and-dropping a large/complex STL file to Cura (at least on Windows) causes a hang in the owner application of the original file browser the drag-and-drop was initiated from while Cura parses and loades the STL.

Ideally, the code in the drag-and-drop handler should return as quickly as possible since it seems to be (again, at least on Windows) a blocking call, and should only register the path to the STL and trigger a load in a separate thread (or even in the main thread).

Of course, Windows Explorer probably shouldn't be making a synchronous call and blocking while Cura loads the file, but there's not much we can do about _that_.

Ghostkeeper, 2017-04-04:
We do load models asynchronously. Any hanging is the result of thread switching overhead from your computer or the transfer of the mesh data to the GUI thread for rendering.

mqudsi, 2017-04-04:
Could the latter operation take too long under certain circumstances, for certain models, etc? I'm not able to reproduce this each time, but under certain circumstances the hang occurs before the loading begins. Ideally, the filename POST'd to the main GUI thread pump on the drag action would be stored and a timer/event/bool set to indicate that a load should happen in the worker thread. 

Is a handle to the STL/3mf file dragged to Cura opened/obtained in the synchronous call? Perhaps if the file takes too long to open (uncached data, external media, etc) that is causing the delay?

awhiemstra, 2017-04-04:
Drag/drop in Cura is mostly handled through Qt, we just respond to the drop event from Qt. This is completely separate from platform code, we never see the platform event. I would expect Qt to deal with the fact that some applications apparently expect a reply to the drop event.

What may be happening is that Qt waits for the client code (our code) to finish dealing with the drop event before sending a reply. During our drop event handling, the code to load the model starts on a separate thread. Due to Python's broken scheduler, this then hogs the GIL, never allowing things to properly return to Qt's event loop to actually send the event. But I believe we actually have some code that makes sure to at least perform a few iterations of the event loop before performing the real load.

mqudsi, 2017-04-05:
That makes sense, and the behavior I was seeing would be explained by a GIL issue. It's rather sporadic and not of major concern; I was only pointing it out in case it was an easy/obvious fix and I wouldn't worry about it too much! :)

BagelOrb, 2017-07-21:
I don't think we need to keep this issue around. Correct me if I'm wrong.

closed 2017-07-21 by BagelOrb

mqudsi, 2017-07-21:
Nope, thanks.