Ultimaker/Cura #1644: Uniform Scaling checkbox rarely works unchecked

mqudsi, 2017-04-03:
I've had this problem since 2.4 beta and I apologize for not reporting it before, but it's still here in 2.5 beta. Basically, it is almost impossible to turn off uniform scaling - though it will randomly work as expected.

I suspect this _might_ be a platform-specific thing or else it would have been reported long ago, so I'm attaching a video showing the bug. Apologies for the horrible watermark smack-dab in the middle.

Also, this video happens to show an example of the "focus is lost while typing" issue.. the "12%" scaling happened while I was in the process of typing in "120" but the focus was lost after entering the first two digits.

Screencast: https://www.youtube.com/watch?v=IjLH_C_gSQk&feature=youtu.be

Ghostkeeper, 2017-04-04:
Could be a duplicate of https://github.com/Ultimaker/Cura/issues/1637

Could you make sure to not press the Ctrl key? I'm not reproducing it unless I press Ctrl.

mqudsi, 2017-04-04:
I think it might be a duplicate indeed.

closed 2017-04-05 by mqudsi