Ultimaker/Cura #1646: Brim Loops ignoring Squish Factor, Overlapping

Shanjaq, 2017-04-04:
I am having a problem with Cura adding brims on all quality levels.  As the brim gets closer to the center, the nozzle isn't moving far enough per loop, so about halfway through the filament is jamming (even with a nice gearhead extruder) because the nozzle is completely covered by the previous loops and it's squirting out the sides.  I can't seem to find any Cura settings to adjust the first-layer line width, which would influence the spacing of the loops as expected.  I set the nozzle diameter to 0.3mm, which is according to the Printrbot Simple specification on their website, and if I don't use a brim the print comes out nicely, if not for the curling...

I cannot raise my extruder any higher because the filament will not stick.  I think it's a software problem, it's not accounting for the flattening of filament on the build plate while advancing toward the center.

I have also tried fiddling with the extrusion multiplier, but it just makes the loops have ragged edges and when it finally gets to the model itself, the lines are too thin and the layers are not sticking together.  I have no printing problems with PrintrBot's cloud slicing, but their brim functionality is broken so I thought I'd try Cura with the manufacturer's recommended settings, which brings me to the problem described in this post.

Ghostkeeper, 2017-04-04:
You could try setting the actual line width in Cura to 0.3 as well.

We default the line width to 7/8ths of the nozzle diameter because we found that the lines adhere better to each other when you do that. If that doesn't work for you then you can adjust the line width. There's a specific setting for Skirt/Brim Line Width as well, under the Quality category (click the gear icon in the category header to make the setting visible).

Shanjaq, 2017-04-05:
Ok, I expanded the first layer line width from 0.3 to 0.55, it's marginally better.  Question:  does setting the line width cause the extruder to work overtime in order to fill those extra 0.25mm with plastic?  or does it only specifically change the spacing?

Ghostkeeper, 2017-04-05:
It adjusts the flow as well as the spacing. In theory it should extrude enough material to fill the new width of the line entirely with plastic.

Practice is a bit different due to pressures in the nozzle and surface tension of the material. Ranges between 50% and 150% of the nozzle width seem to work alright, at least on the Ultimaker machines we have at work.

Shanjaq, 2017-04-06:
Ok, thanks for the insights.  I'll see if PrintrBot is willing to release the secret sauce for their cloud slicer since it works beautifully, if not for the broken support functionality.

closed 2017-04-06 by Shanjaq