Ultimaker/Cura #1652: Cura 2.5 Can't Slice More Than 3 Parts

peterbrown77, 2017-04-05:
When set at one at a time with very fine settings (0.06mm layer height), Cura fails and says that parts are not in the build volume though the definitely are.

https://drive.google.com/open?id=0B5J2gg__lVTNcVhpV3JIaFA5dmc

nallath, 2017-04-05:
You set the size of the head to 240mm. So for every model printed one at the time, Cura tries to put ~24 cm between the objects.

closed 2017-04-05 by nallath

peterbrown77, 2017-04-05:
Thanks! Those were the settings Makergear gave me to use. I have gone back to them to see what the correct Y value should be.... This was driving me nuts!

nallath, 2017-04-06:
They might want to consider creating a proper definition file and making a pull request. That way other users won't have to mess around with custom machine setup.

peterbrown77, 2017-04-06:
The tech just sent me a screenshot of his settings, obviously a typo on his part bc the Y axis is 240 not the print head size...but hey, what do I know hahah?