Ultimaker/Cura #1653: Veiws,

UltimateSports, 2017-04-05:
Hi, Please can it be considered to allow for snapping the view?

to compare to solidworks and other modeling software, arranging parts when the view is snapped to top or front etc can be very useful! especially if more accurate than 'scrolling' the view.

Ghostkeeper, 2017-04-06:
It can be considered (in fact, it has been considered before).

How would you imagine this to work? It snaps to the ordinal directions when you rotate the camera? Or you can click the lines of the axis origin at the bottom left of the build volume, perhaps? Previously clicking on a cube was considered but that was apparently patented by AutoDesk. So we can find a workaround by making the UI different.

UltimateSports, 2017-04-06:
if scrolling close (20deg maybe ) to the perpendicular and a double click of a scroll wheel or tap of the spacebar? the idea of clicking the arrows is good too. 
how close would holding crtl and clicking a face in the printer view be to the AutoDesk patent? too close maybe? 
a shortcut under 'v' could bring up a menu with top, front, left, right. 
or another shortcut being alt and arrow keys. up being top, down being front, right-right, left-left...

ChrisTerBeke, 2017-11-14:
@alekseisasin is working on something like this, maybe he has some insights to share?

alekseisasin, 2017-11-15:
I pushed changes for this ticket: CURA-4527
<https://jira.ultimaker.com:8443/browse/CURA-4527>
The space mouse is only on my local repository, because I didn't want to
commit every not working change

Ghostkeeper, 2017-11-24:
We've added some buttons in Cura 3.1 to move to the ordinal directions. It's not the snapping that you asked for but it might fill the same role. The beta for 3.1 should go online in a few minutes.

closed 2018-06-29 by alekseisasin