Ultimaker/Cura #1654: Cura 2.5 on Ultimaker 3 "Print Finished" but no print

eghterafi, 2017-04-05:
I installed Cura 2.5 and started working with it, but after a day anything i want to print it says "Print Finished. Carefully remove the print..... on my ulimaker 3. 
not i am printing on the wireless network. 
I tried to add new printer and try and had the same problem, the only way is to uninstall Cura and install it again.

nallath, 2017-04-06:
Check your printer. It should display the same message. As long as it's there, the machine reports that it' s still printing.

Ghostkeeper, 2017-04-06:
Hold up. I think Nallath read this issue wrongly.

Eghterafi reports that whenever he starts printing, the printer says that it's done printing without printing anything.

Ghostkeeper, 2017-04-06:
It could be related to the wifi bug we found in the last few days. Or perhaps the user set some setting weirdly, like layer height to 20m, which causes the g-code to be empty.

Could you save a print that has this problem as a project so that we can reproduce this issue? It's under File -> Save Project.

eghterafi, 2017-04-06:
I checked all the settings, Ghostkeeper is correct, the message is on the printer, so i press print on network, and then immediately it says print finished. 
it does not happen on cure 2.4

Ghostkeeper, 2017-04-07:
It doesn't happen for me. I'm not reproducing it here :confused:

Icemixx, 2017-04-21:
Same here. Using LAN. But I found a workaround. It only does the error, if I load a file with double clicking. I clear the build plate, load file using open file menu, and then it works

fieldOfView, 2017-04-21:
Then the problem is probably the an empty Jobname (in the lower right of the viewport). Fill in a name there before printing and it should work.

For the devs: See https://github.com/Ultimaker/Cura/issues/1703

Icemixx, 2017-04-21:
you are perfectly right! I can't start a print now because i already have an active one, but if I load the file, there is no jobname

Ghostkeeper, 2017-04-25:
I'll close this one as a duplicate of #1703 then.

closed 2017-04-25 by Ghostkeeper