Ultimaker/Cura #1657: temps before machine_start_gcode

maukcc, 2017-04-06:
When no temp is done in the start-gcode, you get the temps before the start_gcode in the folowing fashion

> M140 S60
> M104 S215
> M104 T1 S185
> M104 T2 S185
> M104 T3 S185

then at the end of the file you get 

> M104 S0
> M104 T1 S0
> M104 T2 S0
> M104 T3 S0
> ;End of Gcode

The problem here is that the last command in the end is about T3. If I now play this file again without resetting my printer the printer gets the command M104 S215 and assumes it is still talking about T3

The same goes for the end, if the last extruder used was T2, then that 1st command M104 S0 will be about T2 and T0 will never cool.

So T0 must be specified in start and end

BagelOrb, 2017-04-06:
You should put `T0` in your start gcode.

BagelOrb, 2017-04-06:
The problem doesn't hold for the end temp commands. The `Tx` is only left out for the last extruder.

The UM3 doesn't suffer from the bed staying at the heated temp. Perhaps this is something the firmware should fix?

maukcc, 2017-04-07:
> You should put T0 in your start gcode.

the start gcode comes after the temp settings and will not change anything, so that will not work.
If the printer has any other extruder then T0 as active extruder, the print will fail, and  the user will probably have no idea why. This is a small but serious bug so please re-open this issue

> The problem doesn't hold for the end temp commands. The Tx is only left out for the last extruder.

Correct, but as we need our active extruder to be T0 for the next print (see above), we added the "T0" command in our end_code to set the active extruder to T0. Now if the last extruder used is not T0 but T1  the end looks like:
M104 S0
M104 T2 S0
M104 T3 S0
cooling down T0 and leaving T1 hot, resulting in hardware damage. So when issue 1 is dealt with, by adding T0 to the temp then this is solved also. On a side note: it would look cleaner if all temps would be at the end

> The UM3 doesn't suffer from the bed staying at the heated temp. Perhaps this is something the firmware should fix?

The UM3 probably has an end_gcode that handles the bed temp, so do we :)
But we have to assume that there is no end_gcode

BagelOrb, 2017-04-08:
The UM3 start and end gcode is empty.

This is really not a problem from Ultimaker printers.

Actually the UMO original + dual extruder requires temperatures to be met before the start_gcode is output.

For the UM3 the firmware handles that the state before printing and after printing is consistent, no matter what faulty gcode was used before.

We at Ultimaker are probably not going to fix this any time soon.

jackha, 2018-07-18:
Deferred because of inactivity

Ghostkeeper, 2018-08-30:
I think we shouldn't fix this. If some printers need to clean up after their print (e.g. by switching to tool 0) they should put this in their end g-code. There are other printers that specifically don't want to clean up after their print though, because switching to T0 just takes time and they want the printer to do that while waiting for the user to clean the build plate (i.e. indicate earlier that the print is complete). This is exactly what the end g-code is meant for.

We could throw in an M140 S0 command though to cool the build plate. This should incur no additional time.

closed 2023-06-24 by github-actions[bot]