Ultimaker/Cura #1666: cura 2.4.0

jamesjfc123, 2017-04-09:
I have designed a file in sketchup. When I brought it into Cura everything looked the same. I then checked the layers and it changed my model filling in holes that I did not want to be filled in.

fieldOfView, 2017-04-09:
> I have designed a file in sketchup.

There's your problem. Sketchup lets you create meshes that are horrible for 3d printing, because Sketchup was not designed for 3d printing.

You can try to turn on/off some of the mesh fixes in Cura, but your best path towards succesfull 3d printing is learning a different modeling application.

jamesjfc123, 2017-04-09:
Thank you. What software would you suggest I use in place of sketchup.

fieldOfView, 2017-04-09:
I hear good things about TinkerCAD and Fusion360. Both are free for non-commercial work.

fieldOfView, 2017-04-09:
If you feel your question has been sufficiently answered, feel free to close the issue (both of them)

closed 2017-04-10 by nallath