Ultimaker/Cura #1668: Cura OSX crashing on startup

AletheianAlex, 2017-04-09:
Cura 2.4.0 crashes on startup about 2 out of 3 times, with the same crash report and log results every time. I have tried re-installing Cura (including manually removing all pref files and configurations) as well as trying several Mac OS X versions (10.12.3 & 10.12.4) and several different user accounts.   I may have to be downgrade for now to v2.3.1, which hardly ever crashed.  

My details: 2010 Mac Pro with a fresh install of OS X 10.12.4 and a GTX680 cpu with Nvidia web drivers and Qt version Version: 7.7.3.  Here are the crash reports and Cura.log file:

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CURA.LOG:

2017-04-09 14:48:57,800 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-04-09 14:48:57,804 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-04-09 14:48:57,808 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2017-04-09 14:48:57,812 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-04-09 14:48:57,818 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2017-04-09 14:48:57,822 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-04-09 14:48:57,827 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-04-09 14:48:57,832 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2017-04-09 14:48:57,836 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-04-09 14:48:57,841 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PostProcessingPlugin
2017-04-09 14:48:57,857 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-04-09 14:48:57,868 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-04-09 14:48:57,873 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-04-09 14:48:57,877 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-04-09 14:48:57,884 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-04-09 14:48:57,894 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-04-09 14:48:57,901 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-04-09 14:48:57,906 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-04-09 14:48:57,911 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-04-09 14:48:57,918 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2017-04-09 14:48:57,934 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-04-09 14:48:57,945 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UpdateChecker
2017-04-09 14:48:57,945 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-04-09 14:48:57,985 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2017-04-09 14:48:57,992 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2017-04-09 14:48:57,996 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2017-04-09 14:48:58,001 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-04-09 14:48:58,006 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-04-09 14:48:58,011 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2017-04-09 14:48:58,023 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-04-09 14:48:58,028 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-04-09 14:48:58,030 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'external-backend': False, 'disable-textures': False, 'file': []}
2017-04-09 14:48:58,049 - INFO - UM.VersionUpgradeManager.upgrade [100]: Looking for old configuration files to upgrade.
2017-04-09 14:48:58,055 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: Traceback (most recent call last):
  File "/Users/ultimaker/build/2.4/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 310, in _upgradeFile
TypeError: 'NoneType' object is not iterable

2017-04-09 14:48:58,057 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
2017-04-09 14:48:58,058 - WARNING - UM.Logger.logException [69]:   File "/Users/ultimaker/build/2.4/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 310, in _upgradeFile
2017-04-09 14:48:58,060 - WARNING - UM.Logger.logException [69]: TypeError: 'NoneType' object is not iterable
2017-04-09 14:48:58,063 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: Traceback (most recent call last):
  File "/Users/ultimaker/build/2.4/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 310, in _upgradeFile
TypeError: 'NoneType' object is not iterable

2017-04-09 14:48:58,065 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
2017-04-09 14:48:58,067 - WARNING - UM.Logger.logException [69]:   File "/Users/ultimaker/build/2.4/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 310, in _upgradeFile
2017-04-09 14:48:58,068 - WARNING - UM.Logger.logException [69]: TypeError: 'NoneType' object is not iterable
2017-04-09 14:48:58,077 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-04-09 14:48:58,087 - INFO - UM.Backend.Backend.startEngine [71]: Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
2017-04-09 14:48:58,088 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Calling engine with: ['/Applications/Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/Applications/Cura.app/Contents/MacOS/resources/definitions/fdmprinter.def.json', '']
2017-04-09 14:48:58,091 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-04-09 14:48:58,092 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2017-04-09 14:48:58,094 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2017-04-09 14:48:58,096 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-04-09 14:48:58,097 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
2017-04-09 14:48:58,100 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-04-09 14:48:58,104 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-04-09 14:48:58,106 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] (at your option) any later version.
2017-04-09 14:48:58,107 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-04-09 14:48:58,109 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is distributed in the hope that it will be useful,
2017-04-09 14:48:58,110 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-04-09 14:48:58,112 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-04-09 14:48:58,115 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] GNU Affero General Public License for more details.
[... 135 lines cut ...]
81  com.apple.CoreFoundation      	0x00007fff79621b26 __CFRunLoopRun + 934
82  com.apple.CoreFoundation      	0x00007fff79621524 CFRunLoopRunSpecific + 420
83  com.apple.HIToolbox           	0x00007fff78b81ebc RunCurrentEventLoopInMode + 240
84  com.apple.HIToolbox           	0x00007fff78b81cf1 ReceiveNextEventCommon + 432
85  com.apple.HIToolbox           	0x00007fff78b81b26 _BlockUntilNextEventMatchingListInModeWithFilter + 71
86  com.apple.AppKit              	0x00007fff7711ce24 _DPSNextEvent + 1120
87  com.apple.AppKit              	0x00007fff7789885e -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2796
88  com.apple.AppKit              	0x00007fff771117ab -[NSApplication run] + 926
[... cut ...]
99  nl.ultimaker.cura             	0x000000010251c6c0 0x102452000 + 829120
100 nl.ultimaker.cura             	0x00000001024a7eb5 PyCFunction_Call + 165
101 nl.ultimaker.cura             	0x00000001025252e0 PyEval_EvalFrameEx + 25584
102 nl.ultimaker.cura             	0x0000000102529ad2 0x102452000 + 883410
103 nl.ultimaker.cura             	0x000000010251ee71 PyEval_EvalCode + 81
104 nl.ultimaker.cura             	0x0000000102453ae4 main + 1428
105 nl.ultimaker.cura             	0x0000000102453544 start + 52

nallath, 2017-04-10:
It seems like you have a corrupted local folder. Could you try to delete those files and try again?

AletheianAlex, 2017-04-10:
I gave it a go, but it made no difference.  I did a total uninstall/preference-wipe/profile-delete and reinstall again just to be sure.  Finally to test, I did another complete uninstall, trashed prefs and deleted profiles, etc, and reinstalled 2.3.1 and it works fine.  I then trashed everything again and installed the 2.5 beta, and that version worked fine.  Trashed everything yet again and reinstalled 2.4 and it crashes constantly like before.  So for now I will be on the 2.5 beta, and if that acts up in a future patch I'll downgrade to 2.3.1.

ianpaschal, 2018-02-27:
This is fixed in version 3.2.1.

closed 2018-02-27 by ianpaschal