Ultimaker/Cura #1669: Cura Beta (2) 2.5.0 Crashes on Startup

starbuck93, 2017-04-10:
```
Version: 2.5.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "C:\Program Files\Cura 2.5.1\plugins\ChangeLogPlugin\ChangeLog.py", line 44, in getChangeLogString
    logs = self.getChangeLogs()
  File "C:\Program Files\Cura 2.5.1\plugins\ChangeLogPlugin\ChangeLog.py", line 39, in getChangeLogs
    self.loadChangeLogs()
  File "C:\Program Files\Cura 2.5.1\plugins\ChangeLogPlugin\ChangeLog.py", line 61, in loadChangeLogs
    with open(os.path.join(PluginRegistry.getInstance().getPluginPath(self.getPluginId()), "ChangeLog.txt"), "r",-1, "utf-8") as f:
PermissionError: [Errno 13] Permission denied: 'C:\\Program Files\\Cura 2.5.1\\plugins\\ChangeLogPlugin\\ChangeLog.txt'

```

nallath, 2017-04-10:
uuh. Wow. I have no idea how this happend. It tries to load the changelog file (so it can show you what the changes are), but it doesn't have the permissions to do so.

starbuck93, 2017-04-10:
@nallath is it because of all the double slashes? `C:\\Program Files\\Cura 2.5.1\\plugins\\ChangeLogPlugin\\ChangeLog.txt`

nallath, 2017-04-10:
I don't think so. It correctly finds the location, but it's just not allowed to open it.

starbuck93, 2017-04-10:
I only asked because when I typed in that exact URL in Windows Explorer, it pops up an error "Windows cannot find..."

starbuck93, 2017-04-10:
I edited a few permissions of the folder `ChangeLogPlugin` and `ChangeLog.txt` and let a few more entities read/write the file, and now it works.

closed 2017-04-10 by starbuck93